vvip-68/LedPanelWiFi #1: [solved] Wemos D1 mini крэшится при подключении к роутеру

oxyii, 2023-04-22:
Спасибо за поддержку и развитие проекта! Переезд на веб-морду, без сомнения, - лучшее, что могло произойти с проектом!

Собственно, сабж. Долго не мог вычислить причину крэшей... помог дополнительный `DEBUG`, который не нёс в себе никакой смысловой нагрузки, но микроскопическую задержку создал.

вот этот `delay` потерял смысл, потому что после проверки кнопки есть еще один:
https://github.com/vvip-68/LedPanelWiFi/blob/531360bccf0535940fdf7867057395b6820fa0a4/firmware/LedPanelWiFi_v1.14/LedPanelWiFi_v1.14.ino#L653

а вот этот, который "еще один", ставим `delay(1)` и крэши прекращаются:
https://github.com/vvip-68/LedPanelWiFi/blob/531360bccf0535940fdf7867057395b6820fa0a4/firmware/LedPanelWiFi_v1.14/LedPanelWiFi_v1.14.ino#L663

PR сейчас создать не очень удобно, поэтому оставляю в виде issue. За что отдельно прошу прощения.

vvip-68, 2023-04-22:
Спасибо, мне так удобнее. Внесу исправления...

vvip-68, 2023-04-22:
А в каких случаях крашилось то? 
Это процедура подключения к сети WiFi.
Никогда с этим никаких проблем не было...

vvip-68, 2023-04-22:
По документации на ESP8266 модуль, вроде как врот эти пустые delay позволяют ядру выполнять какую-то обработку стека WiFi...

oxyii, 2023-04-22:
По порядку:
При первом запуске контроллер ничего не знает о роутере, создает свою точку доступа, всё прекрасно работает. Через браузер ставлю параметры подключения к роутеру и связь просто теряется. Контроллер уходит в вечный цикл перезагрузок. То есть, он НАЧИНАЕТ попытки подключиться, даже несколько "точек", свидетельствующих об этом, появляется. А потом crash, reboot и по новой. Содержимое порта со стеком ошибки прилагаю.
<details>
  <summary>Порт</summary>

  ```
  WiFiPanel v.1.14.2023.0416-esp8266
  Версия EEPROM: 0xCE
  Host: 'WiFiPanel-0'
  
  Инициализация файловой системы... 
  Использовано 1998848 из 2072576 байт
  
  
  Матрица: плоская 16x16
  Адресация: по подключению
    Угол: левый нижний
    Направление: вправо
    Тип: зигзаг
  
  Доступные возможности:
  + Бегущая строка: шрифт 5x8
  + Кнопка управления: тактовая
  + Синхронизация времени с сервером NTP
  - Управление питанием
  + Получение информации о погоде
  + Управление через Web-канал
  - Управление по каналу MQTT
  - Групповая синхронизация по протоколу E1.31
  - Дополнительный индикатор TM1637
  - Эффекты Jinx! с SD-карты
  - Поддержка MP3 Player
  
  Выделение памяти для LEDS   : 32440 -> 31664 -- 776
  Выделение памяти для OVERLAY: 31664 -> 30888 -- 776
  
  Выбранные эффекты и их порядок: 
     01. Часы
     02. Лампа
     03. Снегопад
     04. Кубик
     05. Радуга
     06. Пейнтбол
     07. Огонь
     08. The Matrix
     09. Шарики
     10. Звездопад
     11. Конфетти
     12. Цветной шум
     13. Облака
     14. Лава
     15. Плазма
     16. Радужные переливы
     17. Павлин
     18. Зебра
     19. Шумящий лес
     20. Морской прибой
     21. Смена цвета
     22. Светлячки
     23. Водоворот
     24. Циклон
     25. Мерцание
     26. Северное сияние
     27. Тени
     28. Лабиринт
     29. Змейка
     30. Тетрис
     31. Арканоид
     32. Палитра
     33. Спектрум
     34. Синусы
     35. Вышиванка
     36. Дождь
     37. Камин
     38. Стрелки
     39. Анимация
     40. Погода
     41. Жизнь
     42. Узоры
     43. Рубик
     44. Звёзды
     45. Штора
     46. Трафик
     47. Слайды
     48. Рассвет
  
  Отключенные эффекты: 
  
  Поиск файлов карт индексов...
     16x16.map
     32x16.map
     48x16.map
  
  Загрузка файла индексов: 16x16.map...
  Загрузка файла карты индексов завершена
  
  
  Найденные анимации:
     Сердце	8x8
     Марио	16x16
     Погода	16x16
  
  Поиск слайдов на FS в папке '16p16'...
  Найдено 48 слайд(oв) размером 16x16:
     airplan.p
     arrow-dn.p
     arrow-lt.p
     arrow-rt.p
     arrow-up.p
     ball.p
     bird.p
     butterfly.p
     clock.p
     comet.p
     cookman1.p
     elka.p
     fish.p
     fox.p
     fox1.p
     frog.p
     frog2.p
     frog3.p
     gift.p
     hand.p
     heart.p
     jinn.p
     kaktus.p
     karate.p
     lebedev.p
     man2.p
     mario.p
     minion.p
     moon.p
     moroz1.p
     mouse.p
     parrot.p
     pigeon.p
     rocket.p
     smile1.p
     smile2.p
     smile3.p
     smile4.p
     smile5.p
     snowman.p
     spiderman.p
     squares.p
     star1.p
     termo.p
     test.p
     weather.p
     winter.p
     zayka.p
  
  Папка '/web' обнаружена.
  Cпискок файлов web-интерфейса:
     [web]
        MI.122e921fbc0a837a.woff2            59000
        [assets]
           [langs]
              eng.json.gz                     7687
              rus.json.gz                     5823
        favicon.ico.gz                         386
        index.html.gz                         1364
        main.js.gz                          325731
        polyfills.js.gz                      11979
        runtime.js.gz                          638
        styles.css.gz                        12981
        wallet.f3b77cb2a7aa9496.png.gz        1169
  Свободно памяти: 29864
  
  Подключение к Router -> 192.168.3.105
  ....
  --------------- CUT HERE FOR EXCEPTION DECODER ---------------
  
  Soft WDT reset
  
  Exception (4):
  epc1=0x401062e8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
  
  >>>stack>>>
  
  ctx: cont
  sp: 3ffffd20 end: 3fffffd0 offset: 0160
  3ffffe80:  3fff20e0 00001950 00000000 3fff17f4  
  3ffffe90:  3fff20e0 00000004 00000000 4020c377  
  3ffffea0:  4024ff6c 08080808 4024ff6c 0103a8c0  
  3ffffeb0:  4024ff6c 00ffffff 4024ff6c 0103a8c0  
  3ffffec0:  4024ff6c 6903a8c0 00000000 40248178  
  3ffffed0:  0000181f 00001240 00002710 00000000  
  3ffffee0:  4024327c 3ffe9613 3fff2258 3fff0415  
  3ffffef0:  3fff0414 3fff17f4 3fff2258 4020c4ad  
  3fffff00:  3fff0414 3fff17f4 3fff2258 40230d62  
  3fffff10:  00000000 00360036 00000000 00000100  
  3fffff20:  00000005 00000020 00000000 0036003f  
  3fffff30:  00000000 00000000 00360036 00000000  
  3fffff40:  00000000 0036003f 00000000 00000000  
  3fffff50:  00363735 00000000 00000000 002d002f  
  3fffff60:  00000000 00000000 fe0020b7 00000000  
  3fffff70:  00000000 0026002f 00000000 00000000  
  3fffff80:  00383438 00000000 00000000 0020002f  
  3fffff90:  00000000 feefeffe feefeffe feefeffe  
  3fffffa0:  feefeffe feefeffe feefeffe 3fff2428  
  3fffffb0:  3fffdad0 00000000 3fff23fc 40246f70  
  3fffffc0:  feefeffe feefeffe 3fffdab0 40100fb9  
  <<<stack<<<
  
  --------------- CUT HERE FOR EXCEPTION DECODER ---------------
  
   ets Jan  8 2013,rst cause:2, boot mode:(3,7)
  
  load 0x4010f000, len 3424, room 16 
  tail 0
  chksum 0x2e
  load 0x3fff20b8, len 40, room 8 
  tail 0
  chksum 0x2b
  csum 0x2b
  v0009c1d0
  ~ld
  ⸮⸮⸮⸮o⸮{⸮⸮g|⸮l⸮$`c⸮⸮|r⸮$⸮g⸮⸮o⸮ d`⸮⸮s⸮$⸮l⸮⸮
  ```
</details>

vvip-68, 2023-04-22:
Понятно. Но странно. Ни на одном из десятков собранных устройств такой картины не наблюдалось ни разу...
Ну ок - помогло -- и замечательно...

closed 2023-04-22 by vvip-68

oxyii, 2023-04-22:
@vvip-68 я и писал тут больше для "истории". Чтобы в поисковиках кто-то мог найти, если с таким столкнется. Еще раз спасибо за поддержку проекта!