vvip-68/LedPanelWiFi #9: нет доступа к настройкам

bdu1977, 2023-10-28:
переехал на новую версию, на 1,14 и тут начались непонятные проблемы. сборка на nodemcu v3, тм1637, dfplayer. поскольку на контроллер уже была запись старой версии, то я поменял #define EEPROM_OK     0xA6 , в Arduino IDE в меню "Инструменты" -> "Erase Flash" выбрать значение "All Flash Contents", загрузил скетч, потом загрузил файловую систему. логин и пароль к роутеру так же были внесены в скетч. далее в телефоне выбираю сеть точки доступа, ввожу 192,168,4,1 и..... нет ничего, страница с настройками не открывается. пробовал отключать передачу мобильных данных, результата нет. подключения к роутеру нет. в мониторе порта пишет что точка доступа активна, но подключиться не получается. при этом все выводится на матрицу и работает сенсорная кнопка. после очередной прошивки смог подключиться к роутеру, в мониторе порта указан адрес панели, но при введении этого адреса в браузер мне пишется, что сервер не активен. при этом на панель ничего не выводится - только первый светодиод моргает красным-желтым-зеленым, хотя в мониторе порта показывается, что идет смена режимов.
Подскажите что у меня не так.
Напишите правильную последовательность загрузки скетча и файловой системы, с указание что тут  нужно менять в Arduino IDE в меню "Инструменты" -> "Erase Flash" выбрать значение "All Flash Contents и что тут. с учетом того что на контроллере уже прошивка была и работала.

vvip-68, 2023-10-29:
Да, вроде, всё правильно сделано.
Если после очередной перепрошивки (да еще и с опцией Erase All Flash Content) пишет, что сервер не найден - значит после заливки скетча вы не залили файловую систему. Она тоже хранится в Flash Content и загрузка прошивки с этим флагом ее, естественно, тоже уничтожает. Горит только первый светодиод - проверьте к тому ли пину подключена лента. Когда зальете файловую систему и удастся открыть Web-интерфейс - назначение пинов можно менять оттуда.

bdu1977, 2023-10-31:
все получилось с....... не знаю какого раза, не понятно почему (смотрел через монитор порта) не все файлы грузились в файловую систему. в конце концов я смог подключиться к матрице, перевести ее на роутер и уже настраивать через web интерфейс. но тут выявилось еще 2 проблемы. 
1. телефон , если хоть раз был подключен к адресу 192.168.4.1 больше не хочет подключаться, там что-то блокируется, буду разбираться.
2. это поведение сенсорной кнопки. управление через web интерфейс идет на ура, но стоит выключить кнопкой и потом включить, то ничего не происходит, мигает только первый светодиод и больше ничего не работает. хотя в мониторе порта все команды переключаются (если управление идет через браузер). если выключить матрицу через браузер, а включить кнопкой, управление через web браузер идет с тормозами, но на панель ничего не выводится. ее можно оживить только выключив питание. 
тут в чем дело?
P.S.
по прошивке 1.13 по бегущей строке можно задать тут вопрос или в отдельную ветку выложить?

vvip-68, 2023-11-01:
По подключению к 192.168.4.1 - а в это время контроллер точку доступа поднял (работает как точка доступа)? Если он подключился к сети WiFi - точку доступа он уже не создает.
По кнопке - странно, но ничего сказать не могу. Даже причин такого поведения представить себе не могу.
По вопросу к 1.13 - лучше там. Вдруг у кого такая же проблема есть или была - или ответит, или кто искать ответ будет - увидит решение там.

vvip-68, 2023-11-02:
Про Web-интерфейс. Версия 1.14 больше рассчитана на ESP32 у которой много оперативной памяти. На ESP8266 работает на небольших матрицах - я проверял до 700 диодов примерно. При бОльшем количестве ей элементарно не хватает памяти. Надо смотреть что пишет в мониторе порта. Для Web-соединения там используется сторонняя библиотека, которая,  возможно, написана не оптимально - она дробит память и со временем доступная большим куском память уже не может быть выделена и сервер, крутящийся на микроконтроллере отказывает клиенту.  На фрагментацию обычно нужно какое-то время, поэтому сразу после перезапуска контроллера оно работает, а через несколько дней может уже и не отвечать. Увы. Версия экспериментальная - она и есть экспериментальная, увы.

В общем - нужно смотреть / погонять что в логе пишет, сколько свободной памяти остается. Обычно, когда не может выделить память на запрос  - контроллер просто перегружался... А так - все подключения и всё отправленное фиксируются с сервера клиенту отображается в мониторе порта...

По сенсорной кнопке Версия 1.13 (на которой кнопка работает) от версии 1.14 отличается только прикрученным веб-интерфейсом и убранным MQTT. Все остальные алгоритмы - обработка кнопки, эффекты и всё прочее - не трогалось вообще.  Поэтому вопрос - там работает стабильно / тут не работает - вызывает недоумение и ответить на него я ничего не могу.

bdu1977, 2023-11-02:
у меня матрица 16*16, так что наверное памяти хватает. сейчас включу на ночь, пусть погоняется, потом весь лог посмотрю. что касается кнопки, то тут полный полтергейст, возможно отключать буду.

vvip-68, 2023-11-02:
Сенсорная кнопка - вещь капризная. Провод до нее должен быть лучше экранированным и коротким. Кнопка питаться от 3.3 вольта, не от +5. Если продолжает ловить непонятно что - напаивается конденсатор 10-30пф на специальную площадку

bdu1977, 2023-11-02:
провод у меня примерно сантимов 20, все три провода свиты в косичку. питание подается 3,3в с отдельного стабилизатора. 
как посмотреть сколько памяти потребляется матрицей, сколько осталось? в мониторе у меня только смена режимов идет, только про память в самом начале было, вот часть лога

WiFiPanel v.1.14.2023.0602
Контроллер: 'NodeMCU'
Версия ядра: ESP8266 v3.1.2
Версия EEPROM: 0xA6
Host: 'WiFiPanel'

Инициализация файловой системы... 
Использовано 1679360 из 2072576 байт

Матрица: плоская 16x16
Адресация: по подключению
  Угол: левый нижний
  Направление:   Тип: зигзаг
Вывод на ленту:
  Линия 1 PIN=D2, START=0, COUNT=256

Доступные возможности:
+ Язык интерфейса: RUS
+ Бегущая строка: шрифт 5x8
+ Кнопка управления: сенсорная PIN=D6
+ Синхронизация времени с сервером NTP
- Управление питанием
+ Получение информации о погоде
+ Управление через Web-канал
- Групповая синхронизация по протоколу E1.31
+ Дополнительный индикатор TM1637 DIO=D5 CLK=D7
- Эффекты Jinx! с SD-карты
+ Поддержка MP3 Player STX=D3 SRX=D4

Выделение памяти для LEDS   : 31232 -> 30456 -- 776
Выделение памяти для OVERLAY: 30456 -> 29680 -- 776

Выбранные эффекты и их порядок: 
   01. Часы
   02. Лампа
   03. Снегопад
   04. Кубик
   05. Радуга
   06. Пейнтбол
   07. Огонь
   08. The Matrix
   09. Шарики
   10. Звездопад
   11. Конфетти
   12. Цветной шум
   13. Облака
   14. Лава
   15. Плазма
   16. Радужные переливы
   17. Павлин
   18. Зебра
   19. Шумящий лес
   20. Морской прибой
   21. Смена цвета
   22. Светлячки
   23. Водоворот
   24. Циклон
   25. Мерцание
   26. Северное сияние
   27. Тени
   28. Лабиринт
   29. Змейка
   30. Тетрис
   31. Арканоид
   32. Палитра
   33. Спектрум
   34. Синусы
   35. Вышиванка
   36. Дождь
   37. Камин
   38. Стрелки
   39. Анимация
   40. Погода
   41. Узоры
   42. Рубик
   43. Звёзды
   44. Штора
   45. Трафик
   46. Слайды
   47. Рассвет

Отключенные эффекты: 

Поиск файлов карт индексов...
   16x16.map
   32x16.map
   48x16.map
Загрузка файла индексов: 16x16.map...
Загрузка файла карты индексов завершена

Найденные анимации:
   Сердце	8x8
   Марио	16x16
   Погода	16x16

Поиск слайдов на FS в папке '16p16'...
Найдено 48 слайд(oв) размером 16x16:
   airplan.p
   arrow-dn.p
   arrow-lt.p
   arrow-rt.p
   arrow-up.p
   ball.p
   bird.p
   butterfly.p
   clock.p
   comet.p
   cookman1.p
   elka.p
   fish.p
   fox.p
   fox1.p
   frog.p
   frog2.p
   frog3.p
   gift.p
   hand.p
   heart.p
   jinn.p
   kaktus.p
   karate.p
   lebedev.p
   man2.p
   mario.p
   minion.p
   moon.p
   moroz1.p
   mouse.p
   parrot.p
   pigeon.p
   rocket.p
   smile1.p
   smile2.p
   smile3.p
   smile4.p
   smile5.p
   snowman.p
   spiderman.p
   squares.p
   star1.p
   termo.p
   test.p
   weather.p
   winter.p
   zayka.p

Папка '/web' обнаружена.
Cпискок файлов web-интерфейса:
   [web]
      MI.122e921fbc0a837a.woff2            59000
      [assets]
         [images]
            esp32.png                       7978
            nodemcu.png                    10874
            wemos.png                      12207
         [langs]
            eng.json.gz                     8596
            esp.json.gz                     8844
            rus.json.gz                     6712
      favicon.ico.gz                         386
      index.html.gz                         1413
      main.e44c7c6a45b2d807.js.gz         337831
      pf.dcddd532a07c6b2d.js.gz            11979
      runtime.7882e08c7503fb34.js.gz         638
      styles.7e4222ebe23cabfb.css.gz       13020
      wallet.f3b77cb2a7aa9496.png.gz        1169

Инициализация MP3-плеера...
SD-карта, готова
Поиск файлов на карте плеера...

Звуков будильника найдено: 20
Звуков рассвета найдено: 7
Звуков сообщений найдено: 21
DFPlayerError: Таймаут!
SD-карта, готова
Поиск файлов на карте плеера...

Звуков будильника найдено: 20
Звуков рассвета найдено: 7
Звуков сообщений найдено: 21
Поиск файлов на карте плеера...

Звуков будильника найдено: 20
Звуков рассвета найдено: 7
Звуков сообщений найдено: 21
Свободно памяти: 26000

Подключение к zveri -> 192.168.1.55
Двойное нажатие на кнопку для прерывания подключения
и переход в режим точки доступа
..............
WiFi подключен. IP адрес: 192.168.1.55

UDP-сервер на порту 2390
HTTP server started
Свободно памяти: 23864
Переход в авторежим отключен
Режим: The Matrix
NTP-сервер ru.pool.ntp.org -> 162.159.200.1
Отправка NTP пакета на сервер ru.pool.ntp.org

Запрос текущей погоды
Погода получена: Yandex
Город: Удельная
Сейчас: пасмурно, +4ºC
Код иконки: 'ovc'
Цвет неба: '#2c4a81'
Сейчас: 
Рассвет: 07:34
Закат: 16:48

UDP пакeт размером 48
UDP << ip='162.159.200.1:123 ntp sync
Разбор пакета NTP
Секунд с 1970: 1698955989
Получено время: 02.11.2023 20:13:09
--------------------
web -> '/web/main.e44c7c6a45b2d807.js.gz' -> send (text/javascript)
FM: 19648 - 18504
web -> '/web/MI.122e921fbc0a837a.woff2' -> send (font/woff2)
FM: 17608 - 15744
web -> '/web/assets/langs/rus.json.gz' -> send (application/json)
FM: 13696 - 12552
web -> '/web/wallet.f3b77cb2a7aa9496.png.gz' -> send (image/png)
FM: 10520 - 9376
WebSocket клиент #1 с адреса 192.168.1.33
WEB пакeт размером 8
<-- $13 2 0;
FM: 19232  Max: 12848  Frag: 32
WEB пакeт размером 62
<-- $6 7|W|H|PS|LG|VR|HN|QZ|FS|SX|CH|CV|WZ|AL|SM|LE|UP|FM|MC|MX|MZ
FM: 18488  Max: 12848  Frag: 30
WEB пакeт размером 31
<-- $6 7|PS|BR|DM|RM|FV|PD|IT|EF|EN
FM: 18272  Max: 12848  Frag: 28
WEB пакeт размером 8
<-- $13 2 1;
FM: 17688  Max: 12848  Frag: 26
WEB пакeт размером 7
<-- $6 7|FV
FM: 17632  Max: 12848  Frag: 27
WEB пакeт размером 8
<-- $13 2 2;
FM: 18008  Max: 12848  Frag: 28
WEB пакeт размером 8
<-- $13 2 3;

vvip-68, 2023-11-02:
>как посмотреть сколько памяти потребляется матрицей, сколько осталось? в мониторе у меня только смена режимов идет, только про память в самом начале было

Ну вот это в логе `FM: 18008` - и есть количество свободной памяти - Free Memory - 18008 байт
Максимальный цельный фрагмент  - Max 12848 байт. Процент фрагментации памяти - Frag: 28%
Все это по результатом того, что  отдает на запрос ядро.

vvip-68, 2023-11-02:
Обычно проблемы начинаются с Web-мордой когда свободной памяти остается 7-8 килобайт, нет одного большого куска такого размера и процент фрагментации - за 80...

bdu1977, 2023-11-03:
т.е. у меня со свободной памятью все нормально?, ну тогда пусть остается так как есть.
спасибо.

vvip-68, 2023-11-03:
Она может дробиться во время работы, так что цельного большого куска может и не оказаться. Но тут я ничего поделать не могу. Наблюдайте...

closed 2023-11-08 by vvip-68