vvip-68/LedPanelWiFi #10: ttp223 и TM1637

bdu1977, 2023-11-04:
версия прошивки 1.14, nodemcu 3v, кнопка ttp223 и индикатор ТМ1637. загрузка прошла хорошо, кнопка ведет себя неадекватно, при каждом обновлении прошивки, без файловой системы. у кнопки разное поведение, при ее нажатии матрица может повиснуть, может начать моргать, но чаще всего просто регулирует яркость, т.е. одно нажатие это уменьшение яркости от мах до min и наоборот, тройное нажатие включает ночные часы и убирает яркость. установка конденсатора не помогает, замена на другую такую же кнопку то же не помогает. Так же на индикаторе всегда горит значение яркости br99 числа разные в зависимости от яркости. значение часов не выводятся на индикатор. но стоит в рабочей матрице отключить кнопку (у меня она на разъеме) на индикатор выводится значения часов, подключаю кнопку-сразу выводится значение яркости. так же сам индикатор частенько помаргивает, такое ощущение, что на долю секунды на индикатор выводится другая информация.

vvip-68, 2023-11-04:
Ну то есть как факт - проблема в кнопке. Как только ее подключаете - начинаются "чудеса". То что постоянно горит br99 и значения часов не выводятся на индикатор - означает, что прошивка все время видит кнопку "нажатой" - на ее выводе всегда присутствует высокий уровень. Светодиод на кнопке постоянно горит?  В скетче то кнопка обозначена как сенсорная? Может у вас кнопка перемычками AB назначена как триггерная, а не отпустил/нажал. Типа один раз нажал - она включилась. Второй раз нажал - она выключилась...

bdu1977, 2023-11-04:
кнопка в скетче сенсорная установлена, перемычек на кнопке не стоит, светодиод не горит. светодиод горит только тогда когда к ней прикасаешься. работа похожа на триггер, но вот что по факту. на индикаторе br99 я прикасаюсь к кнопке, на кнопке загорается светодиод, отпускаю кнопку, светодиод гаснет, а индикация яркости продолжает идти к нулю, если во время понижения яркости, коснуться кнопки, светодиод загорается, отпускаю кнопку, светодиод гаснет, а яркость нарастает.
если просто тупо тыкать по кнопке, зажимать ее и просто касаться и отпускать, то можно вывести на индикатор часы, но при выключении матрицы и последующем включении, индикатор переключается на показание яркости.
вариант того что кнопка неисправна, немного под сомнением, поскольку именно данная матрица, без каких-либо модификациях на прошивке 1.13 работает идеально. и именно эта кнопка работает просто отлично. я пробовал библиотеки поменять, но компиляция не шла, пробовал код подменить в прошивке, но он не компилировался, немного по другому написан. может есть еще какие соображения?

vvip-68, 2023-11-04:
1.13 от 1.14 отличается выкинутым MQTT и добавленным Web-интерфейсом. Все прочие алгоритмы - обслуживание кнопок, эффекты, часы, TM1637, календари и прочий функционал "ядра" прошивки - без изменений.

bdu1977, 2023-11-04:
ну вот что есть, то есть. я не программист, но буду пробовать разные способы, может что-то получится.

vvip-68, 2023-11-05:
Разбирайтесь. Смотрите настройки - тот ли тип кнопки выбран на странице настроек в WebUI. К тому ли пину подключена кнопка. Только что собрал стенд с сенсорной кнопкой и индикатором - всё работает без каких-либо проблем.

https://drive.google.com/file/d/1Pm9Wf8KpeM3N1vbv5NpARkyZHrKYRZp6/view?usp=sharing

[image]

closed 2023-11-06 by vvip-68