vvip-68/LedPanelWiFi #11: stm32

bdu1977, 2023-11-04:
хочу переделать данный проект с заделом на будущее, что бы подключать несколько матриц в наборный экран или использовать матрицу большего размера. можно данный проект исполнить на stm32? есть на данный контроллер какие-то ограничения или другие нюансы по прошивке и обновлению прошивки?

vvip-68, 2023-11-04:
Ну, вроде как stm32 вообще имеет свое SDK и оно не совместимо с SDK Espressif. Для нее вроде даже своя IDE используется - что-то типа STM Cube. Ну то есть скетч переписывать полностью с нуля.

Так а эта прошивка вроде как умеет и матрицы большого размера, и сборные экраны различной конфигурации (даже разных размеров). И даже параллельный вывод на несколько линий - реализовано до 4, но можно дописать, чтобы до 8... Для бОльшего экрана нужна ESP32 у которой больше памяти.

Ну а так - освоить программирование под новый контроллер со своим SDK - хорошее дело, почему бы и нет.

bdu1977, 2023-11-04:
эх, купил одну такую красивую плату, вот валяется бестолку.  Я почему спросил, Алекс Гайвер в самом начале разработки данной гирлянды писал что его прошивку можно на разные платформы прошить. вот я и подумал, а почему бы не использовать данный контроллер, все равно валяется, да и проектов на них почти нет.

vvip-68, 2023-11-04:
И у него при этом в коде была куча ветвлений - вот это - для  вемоса, вот это - для stm
Чего проще - установить ядро STM в ардуино иде, выбрать плату и нажать "скомпилировать". Скомпилится без ошибок - попробовать залить и посмотреть что получится.  Но 99% что не скомпилится и для исправления ошибок потребуется переписать минимум 50% скетча. Есть желание - вперед!

vvip-68, 2023-11-04:
Даже по сути - написание в Arduino IDE и SDK для ESP8266 / ESP32 - по сути костыли (хотя и работают). Потому что родная среда - использование FreeRTOS и родного SDK от Espressif, а все вот это в ардуино иде - по сути костыли / обертка для вызова родного SDK с заведомым урезанием функционала, неоптимальными решениями - лишь бы хоть как-то шевелилось и не падало...

bdu1977, 2023-11-04:
понял. спасибо за пояснение. придется куда-то применить в более простом проекте, что бы не пропадало.

closed 2023-11-04 by bdu1977