vvip-68/LedPanelWiFi #18: Ссылка на ленту для гирлянд задублирована в About

saippuakauppias, 2023-11-11:
Судя по всему, ссылка на 12см ленту некорректная – там лента с лаковым покрытием проводов.

Случайно наткнулся тут: https://github.com/vvip-68/LedPanelWiFi/blob/main/wiki/About/about.md

```
Ленту с расстоянием между диодами 12 см можно купить [тут](https://aliexpress.ru/item/32707248386.html):

Ленту с расстоянием между диодами 10 см можно купить [тут](https://aliexpress.ru/item/32707248386.html):
```

И ленту с 12см никак не могу найти на алике сейчас, а все что по 10см – на части фото в отзывах с нормальным покрытием, а на других – с лаковым (такое что-то брать совсем уж не хочется, тем более она скорее всего не гибкая будет).

vvip-68, 2023-11-11:
Ну что делать - ссылки на али долго не живут, меняются продавцы, меняется ассортимент товалров. Но направление мысли то понятное. Теперь придется самому искать на Али. Ну, возможно 12-сантиметровые вообще сняли с производства. Не знаю...

closed 2023-11-11 by vvip-68

7918514, 2023-11-11:
Всем привет. Наткнулся на интересные ленты длиной аж до 200метров и с разным расстоянием между диодами. Сам такие не покупал, про качество не скажу.
https://aliexpress.ru/item/1005004633923201.html
https://aliexpress.ru/item/1005005594095201.html

saippuakauppias, 2023-11-11:
Первые две на WS2811, такие кажется не подойдут :( 

А другие WS2812, что вроде бы должно быть ок (но я парой часов раньше купил на основе WS2812B IC у другого продавца, где каждые 10см диоды), вот эти: https://aliexpress.ru/item/1005004846421075.html

Хотя в целом выглядит заманчиво купить бобину 100м, где будет расстояние по 5см, для моей идеи (1.2м х 4 м) будет вообще шикарно смотреться, правда БП туда не ясно какой брать)) В описании товара написано 1 диод = 0.075Вт, получается что:
100м / 5 штук каждый см = 2000 диодов
2000 диодов * 0,075 Вт = 150 Вт
150 Вт / 5 В = 30 А

если ничего не путаю в расчетах и если один диод реально 0.075Вт всего, потому что если он 1 Вт, то уже получится 400 А

7918514, 2023-11-11:
WS2811 это чип, а сами диоды WS2812. У китайцев постоянно путаница с этим.
По поводу мощности: у меня панель на 1050 диодов в пике потребляет 7А. Запитывать конечно нужно с обоих концов и ещё в середине и середине середины... 100 метров тонкого провода не шутки))

saippuakauppias, 2023-11-11:
Если там действительно такое малое потребление, то это очень круто) Я видимо перебдел и заказал на 60А БП, но ничего – если всё пойдет по плану, то у меня забор длинный – ещё лент закажу потом)))

А у вас панель на чипах WS2812 или *b? Потому что судя по гуглу – WS2812b должны быть по многим параметрам лучше, думаю что на большом количестве диодов это должно быть заметно

7918514, 2023-11-12:
У меня панель вот на таких диодах. 
https://disk.yandex.ru/d/fIo2UEuKpR54hg
Смотрите в папке **Гирлянда v2**

saippuakauppias, 2023-12-14:
@7918514, заказал одну из этих лент в итоге (2шт, по 50м с расстоянием между диодами в 5см) – вроде как всё работает отлично, напряжения до конца ленты (50м) само собой не хватает и белый свет диодов превращается под конец в тусклый красный)) Но эффекты кажется что работают нормально, цветовая схема такая же и всё кажется ок. 
Надеюсь скоро соберу 4м на 1.2м гирлянду из почти 2к диодов))) Лишь бы esp32 пережевал это и не терял сигнал под конец, чтобы эффекты выглядели хорошо (или буду на несколько "линий" разводить, пока не пробовал вместе такую длину)

vvip-68, 2023-12-15:
Я на esp32 просто для тестов гонял 128x128 матрицу - то есть 16384 диодов. Физически, конечно, было подключено меньше, но данный размер позволял проверить хватает ли памяти и - сигнал все равно формирует пакет на все диоды. Ну да, работало медленнее. Распараллеливание даст результат, будет более шустро выводить. Но работало...

Тут кто-то писал, что делал 3 сегмента (кажется тоже на 2000 диодов). Наблюдал, что на трех сегментах были какие-то артефакты, а когда поделил на два сегмента - артефактов не было. Но тут уже вопросы к библиотеке FastLED. Это в ней реализовано распараллеливание вывода. Даже если там есть ошибки - я с этим ничего поделать не смогу.

По питанию матрицы и падению напряжения - производитель лент вообще рекомендует подавать дублирующее питание через каждые 100 диодов. Заводские матрицы 16x16=256 диодов нормально работают при подаче питания с одного конца... Тут наверное вопрос экспериментальный. Какова общая матрица, какова толщина проводов питания линейки / ленты / струны светодиодов...

saippuakauppias, 2023-12-15:
> Я на esp32 просто для тестов гонял 128x128 матрицу - то есть 16384 диодов
Вау) Ну тогда всё точно должно завестись, у меня были сомнения лишь в том, что esp32 вывезет такое количество диодов))

Подключать питание – да, буду смотреть по факту как будет, но планирую каждые 160 диодов пока что, возможно когда соберу каркас – чуть подкорректирую это (чуть реже сделаю), но скорее всего особого смысла реже делать нету и лучше чаще наоборот, все равно придется ленту резать и удлиннять на местах сгибов, чтобы она ровнее легла и не было перекосов.

За совет с 2 линиями за место 3 – спасибо! Интересная штука, как раз тоже думаю придется докручивать это уже по месту.

saippuakauppias, 2023-12-21:
Привет!
Думаю, стоит оставить для истории тут коммент)
Всё собрал в итоге, вышло 80х24 = 1920 диодов, лента длинной 100м. Подключал питание на каждые 2 ряда (через 160 диодов), использовал 60А БП (скорее всего его даже слишком много). Сделал на 1й линии всего (ESP32 если что), все работает отлично! Думал будут задержки на последних рядах, но нет - сигнал доходит быстро и скорее всего даже если они есть – ничего не заметно для глаза. В итоге решил на 2 линии не разделять :) 

PS: @vvip-68, можете ссылку на спонсорство в репе добавить помимо paypal ещё, например, тинькоф/сбер или yoomoney? Очень хочу задонатить за то, что поддерживаете и дорабатываете этот прекрасный проект!

vvip-68, 2023-12-21:
paypal - остался рудимент, получить с него все равно ничего невозможно.
Ссылка на сберкарту там есть.
Тинькова или юмани у меня нет.