vvip-68/LedPanelWiFi #21: Compilation error: ESPAsyncWebServer.h: No such file or directory

himvampir, 2023-11-14:
Очень прошу помогите. делаю детям гирлянду первый раз в жизни использую ардуино и тп. 
не могу скомпилировать код. Вероятно где-то допустил ошибку но понять не могу. 
In file included from C:\Users\bot_h\Desktop\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\LedPanelWiFi_v1.14.ino:153:
C:\Users\bot_h\Desktop\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\a_def_hard.h:461:10: fatal error: ESPAsyncWebServer.h: No such file or directory
  461 | #include <ESPAsyncWebServer.h>   // Библиотека асинхронного Web-сервера
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: ESPAsyncWebServer.h: No such file or directory

himvampir, 2023-11-15:
> 1. Установить в ардуино иде ядро требуемой для проекта версии
> 2. Скопировать в папку библиотек (обычно в Мои документы/Arduino/Libraries библиотеки из папки Libraries проекта
> 3. Перезапустить Arduino IDE и выполнить сборку проекта
> 
> [https://github.com/vvip-68/LedPanelWiFi/wiki/Пошаговая-инструкция-подготовки-среды-и-сборка-проекта](https://github.com/vvip-68/LedPanelWiFi/wiki/%D0%9F%D0%BE%D1%88%D0%B0%D0%B3%D0%BE%D0%B2%D0%B0%D1%8F-%D0%B8%D0%BD%D1%81%D1%82%D1%80%D1%83%D0%BA%D1%86%D0%B8%D1%8F-%D0%BF%D0%BE%D0%B4%D0%B3%D0%BE%D1%82%D0%BE%D0%B2%D0%BA%D0%B8-%D1%81%D1%80%D0%B5%D0%B4%D1%8B-%D0%B8-%D1%81%D0%B1%D0%BE%D1%80%D0%BA%D0%B0-%D0%BF%D1%80%D0%BE%D0%B5%D0%BA%D1%82%D0%B0)

я прошу прощения я вроде все по инструкции делал, а подскажите есть ли возможность чтоб вы мне помогли удаленно, я Вам оплачу естественно. 
Если да то напишите мне в телеграмм пожалуйста @himvampir

himvampir, 2023-11-15:
и еще вопрос  все завелось вроде подключаюсь к точке доступа а вместо вебморды пишет сервер не активен это что нужно сделать?

vvip-68, 2023-11-15:
Прочитать в wiki проекта раздел с FAQ именно с таким названием
--
Отправлено из Mail.ru для Android четверг,

closed 2023-11-19 by vvip-68