vvip-68/LedPanelWiFi #36: Led Panel WiFi

romchik1982, 2023-12-08:
Уважаемый автор, спасибо за отличную программу! Много всяких настроек и возможностей, но неужели нет такой простой опции как сделать текст бегущей строки жирным?

vvip-68, 2023-12-08:
Простой опции? Ну-ну...
Буквы в шрифте - битовые карты (картинки по факту) букв. Определены в файле fonts.h
Рисуйте свои шрифты как вам нравится. Плюс немного переписать процедуру вывода буквы, т.к. сейчас размеры (ширина одной буквы) - для "тонких букв". Рисуйте, переписывайте.

Учтите также, что все эти массивы хранятся в памяти постоянной микроконтроллера, которой не слишком много. Так что хранить одновременно шрифты и узких и широких букв у вс скорее всего не хватит памяти...

closed 2023-12-10 by vvip-68

saippuakauppias, 2023-12-15:
@romchik1982 , если бегущая строка нужна без макросов, то проще всего сделать нужный размер текста через ПО «Jinx!» (есть в репозитории). Там есть Input text или как-то так эффект. В сети на ютубе множество примеров использования этой программы. Даже есть туториалы как гифки конвертить через неё (без необходимости записи экрана).

@vvip-68, а как "выделять нужно цветом"? Я с бегущей строкой пока не разбирался, потому что собирал макет на коленке, но не помню такой настройки в веб-интерфейсе. Оно там или в самом коде где-то задается?

vvip-68, 2023-12-16:
@saippuakauppias  это в макросах бегущей строки. Макрос {C}
https://github.com/vvip-68/LedPanelWiFi/wiki/Настройка-бегущей-строки

romchik1982, 2023-12-17:
Мне как раз таки макросы устраивают, удобно писать всякий текст на окне для рекламы! Но хотелось бы как то, чтобы буквы были не по одному пикселю, а хотя бы в два, ну чтобы текст был более жирным! Я использую версию 1.13
Я смотрел макрос (С) но там только цвет и все, больше я ничего не нашел В описании макросов тоже что то про жирный текст ничего не сказано!
--
Отправлено из Mail.ru для Android пятница,