vvip-68/LedPanelWiFi #41: [web] Выставлять максимальное число диодов из матрицы после сохранения размеров

saippuakauppias, 2023-12-14:
В случае если меняется размер матрицы в сегментах
[image]

При записи значения в память – не происходит изменение количества диодов на линии:
[image]
Хотелось бы, чтобы здесь записывалось автоматом максимальное значение (ширина х высота) диодов.

Я столкнулся с этим, когда тестировал разные размеры диодной ленты: менял ширину и высоту, но количество диодов на линии не увеличивалось само (я и не смотрел туда), поэтому думал что не хватает напряжения. Но только после того как разрезал ленту и воткнул доп питание на другой конец – осознал, что что-то тут не так и нашел этот кейс :)

Сходу, если честно, не очевидно. Тут как минимум бы пояснение комментарием нужно, а в идеале - записывать сразу туда максимальное значение диодов (если выбрана всего 1 линия) после сохранения.

PS: спасибо за поддержку проекта! Он крутой и вышел вообще офигенным с этим web-интерфейсом новым)

vvip-68, 2023-12-15:
Там же на странице установки размеров матрицы явно написано - что это нужно сделать... Что не очевидно?
Шрифт предупреждения нужно сделать покрупнее? Цвет с синего на оранжевый поменять?

Установка автоматом - сомнительна. Ну да, в большинстве случаев - одна линия и так оно и есть. Но есть масса частных случаев - когда линии, например, две.  И размер матицы распределяется между пинами. Когда несколько матриц в синхронизации и мне нужно выводить иметь полный массив в указанные размеры матрицы, но на подсоединенный к ЭТОМУ контроллеру "экран" выводить не всю матрицу, а только частично. И так далее.

[image]

saippuakauppias, 2023-12-15:
Вот блин, а его я вообще не заметил) Удивительно

closed 2023-12-16 by vvip-68

saippuakauppias, 2023-12-16:
PS: лучше оранжевый сделать, да)

И по поводу линий: если установлена 1 линия, то кмк можно на всю эту линию максимум и давать. А если уже 2 или 3 - то там пусть остается как есть

vvip-68, 2023-12-16:
В последней версии я сделал оранжевым.
По пинам. Вот текущая ситуация. У меня 4 окна. На каждое окно - по гирлянде. Каждая со своим контроллером. Один мастер + 3 приемника. У каждого по одной линии. Контроллер имеет полную матрицу на изображение. Приемники - принимают "вырезки" с этой общей большой матрицы. И хотя полный размер у меня 45x10 - 450 диодов на контроллер нужно выводить только первые 100 диодов. Зачем я буду туда автоматом прописывать все 450?

saippuakauppias, 2023-12-16:
Хм, понял. Поинт весомый и понятный, согласен)