vvip-68/LedPanelWiFi #45: Не открывается WEB интерфейс

altdaps, 2023-12-19:
Прошиваю ESP32_DEVKit, заливаю через SPIFS файлы, подключаюсь через точку доступа PanelAP, захожу по 192.168.4.1, настраиваю подключение к своей wifi сети, после этого достучаться до web интерфейса по ip адресу, выданному по dhcp не могу. Лог показывает, что подключение есть, показывает ip адрес, происходит синхронизация времени через ntp, web интерфейс роутера тоже показывает клиента esp32. Что я делаю не так?

vvip-68, 2023-12-19:
А что пишет лог браузер на вкладке сеть когда в нем F12 нажимаешь?

altdaps, 2023-12-19:
Не совсем понимаю как это работает, но по адресу 192.168.0.100 открывается интерфейс, хотя адрес выдается 0.107. 
В коде есть такая строчка #define DEFAULT_IP {192, 168, 0, 100}, что она дает?

vvip-68, 2023-12-19:
Эта строчка прописывает IP адрес, которое получит устройство после прошивки скетча на ЧИСТУЮ память. Это  значение сохраняется в EEPROM и при последующих загрузках, если не прошивать с очисткой памяти - адрес берется уже оттуда. Его можно изменить через Web-интерфейс.

Можно ставить 0,0,0,0  - тогда адрес будет получаться через DHCP, но я тогда не представляю как вы будете искать устройство в сети, чтобы открыть его WebUI/ Каждый раз лезть в роутер и смотреть куда на этот раз село устройство? Так себе удовольствие...

closed 2023-12-28 by vvip-68