vvip-68/LedPanelWiFi #46: Нет высокого уровня

AHgPEu1, 2023-12-21:


AHgPEu1, 2023-12-21:
Недавно решил подключить к данной гирлянде реле на MOSFET и обнаружил одну проблему, на пине D1 отсутствует высокий уровень (HIGH), хотя в скетче включен USE_POWER 1 и POWER_ON  HIGH. В web-настройках тоже пробовал и так и так, ничего не меняется, на пине D1 низкий уровень - GND. Пробовал задействовать пин D0, на нем тоже становится низкий уровень. Сама плата WEMOS исправна. Когда назначаешь пин D0, например, на D1 сразу высокий появляется. Т.е. пин исправный.  И когда время в web-настройках задаешь время, допустим включить какой-то режим, на пине не появляется высокий уровень. Ощущение, что где-то в коде ошибка.

vvip-68, 2023-12-21:
В коде нет ошибки. У меня к этому пину светодиод припаян на тестовом стенде. И все загорается при включении и гасне при выключении.

AHgPEu1, 2023-12-21:
[image]

AHgPEu1, 2023-12-21:
> В коде нет ошибки. У меня к этому пину светодиод припаян на тестовом стенде. И все загорается при включении и гаснет при выключении.
В какую сторону копать? Т.е. высокий уровень на пине должен появляться просто при включении платы? Кстати, в таймере в web интерфейсе нет слова ВКЛЮЧИТЬ, только ВЫКЛЮЧИТЬ. Я так понимаю, что включить - это DEMO режим?

vvip-68, 2023-12-21:
Да, при включении платы (точнее когда в loop() входит) просто должен на пин подаваться высокий уровень
Это в файле a_main.ino примерно здесь:

[image]

А что за таймер в веб-интерфейсе? И ВКЛЮЧИТЬ или ВЫКЛЮЧИТЬ?
В режимах по времени - то что "Выключить матрицу" - это ВЫКЛЮЧИТЬ, всё остальное - ВКЛЮЧИТЬ

[image]

vvip-68, 2023-12-21:
[image]

https://drive.google.com/file/d/1YSsT8gimIm-9QuzqltwGActiW_72-TOg/view?usp=sharing

AHgPEu1, 2023-12-21:
Ничего не помогает, нет и всё высокого..

WiFiPanel v.1.14.2023.0602
Контроллер: 'Wemos d1 mini'
Версия ядра: ESP8266 v3.1.2
Версия EEPROM: 0xA5
Host: 'WiFiPanel'

Инициализация файловой системы... 
Использовано 655360 из 2072576 байт

Матрица: плоская 15x12
Адресация: по подключению
  Угол: левый нижний
  Направление:   Тип: зигзаг
Вывод на ленту:
  Линия 1 PIN=D2, START=0, COUNT=180

Доступные возможности:
+ Язык интерфейса: RUS
+ Бегущая строка: шрифт 5x8
- Кнопка управления
+ Синхронизация времени с сервером NTP
+ Управление питанием PIN=D1
+ Получение информации о погоде
+ Управление через Web-канал
- Групповая синхронизация по протоколу E1.31
- Дополнительный индикатор TM1637
+ Эффекты Jinx! с SD-карты CS=D8, CLK=D5, MISO=D6, MOSI=D7
- Поддержка MP3 Player

Выделение памяти для LEDS   : 28696 -> 28152 -- 544
Выделение памяти для OVERLAY: 28152 -> 27608 -- 544
и т.д.

vvip-68, 2023-12-21:
Ну, не знаю тогда. У вас версия еще июньская. Возьмите свежую - может чего поменяется...

AHgPEu1, 2023-12-21:
> Ну, не знаю тогда. У вас версия еще июньская. Возьмите свежую - может чего поменяется...

Ёоо... На новой прошивке всё заработало как часы)))

p.s. Может заодно подскажите, как сделать, чтобы эффекты с SD-карты можно было вызывать в разных местах "выбора эффекта"? Пробовал создать в коде еще одно название "SD-карта"- это не помогло. Вторая оказалась пустышкой. Хотел сделать дубль SD-карты и вторую вызывать в другом месте эффектов)

vvip-68, 2023-12-21:
А в чем смысл?

AHgPEu1, 2023-12-21:
> А в чем смысл?

Допустим, на SD-карте находятся 5 эффектов, а всего в DEMO режиме 40 эффектов. И каждый эффект с SD-карты будет повторяться только через 40 эффектов, если включить на ней последовательный выбор, так ведь? Или программа все эффекты с SD-карты один за одним покажет?

vvip-68, 2023-12-22:
А если поставить рандомный выбор - то эффекты на карте выпадают примерно в 10 раз чаще чем прочие эффекты. Каждый раз со случайным роликом. Но есть режим в настройках - последовательно - когда с SD карты ролики показываются последовательно один за одним, пока не закончатся, после чего переходят к обычным эффектам и далее последовательно, пока не вернется к SD-карте, все ролики с нее и так далее...

AHgPEu1, 2023-12-22:
> А если поставить рандомный выбор - то эффекты на карте выпадают примерно в 10 раз чаще чем прочие эффекты. Каждый раз со случайным роликом. Но есть режим в настройках - последовательно - когда с SD [...]

Спасибо.

closed 2023-12-22 by vvip-68