vvip-68/LedPanelWiFi #47: Возможно ли добавить в прошивку MQTT?

GeorgyBenelli, 2023-12-22:
Здравствуйте.Правильно ли я понимаю что в 1.14 убрали mqtt?В a_def_soft.h нет этих настроек.Возможно ли вернуть MQTT?Спасибо

vvip-68, 2023-12-22:
Да, убрали. Об этом даже на первой странице проекта написано.
Вернуть можно. Дело за малым - никто не может сформулировать что нужно и как это должно работать.

Нет постановки задачи, нет понимания как это правильно организовать, какие функции должны присутствовать, нет детальной проработки -> нет реализации.

GeorgyBenelli, 2023-12-22:
> Да, убрали. Об этом даже на первой странице проекта написано. Вернуть можно. Дело за малым - никто не может сформулировать что нужно и как это должно работать.
>
> Нет постановки задачи, нет понимания [...]

Да хотяб как в прошлой версии)В основном задача легкой интеграции в различные системы умного дома.А если по существу то хотелось бы управлять/получать статус по MQTT хотяб:
1.Вкл/выкл
2.Эффект
3.Пресет
4.Яркость
5.В идеале текст слать по Mqtt и чтоб он на панели был(Например с телеграм;)
Ну в общем хотелось бы хотя бы базовые функции видеть в MQTT.

Спасибо за отклик

vvip-68, 2023-12-22:
В прошлой версии задача MQTT выполняла на 90% обслуживание связи андроид-приложения для управления вне локальной сети. И все команды (формат команд) были заточены исключительно под это приложение. То как было реализовано там - это во многом противоречит нормальной, классической реализации связи по MQTT. Что вызывало многочисленные трудности и нарекания от тех, кто пытался построить на этом интеграцию с умным домом.

Приложения больше нет и в этом смысле MQTT перестал быть нужным - я убрал его.

То что вы говорите - это общие слова и пожелания без определения технической реализации. На этом сод не напишешь и связь не построишь. Я понятия не имею как работают умные дома и как технически все это должно быть реализовано. Поэтому и не берусь за это дело.

И похоже что те, кому нужно управление по MQTT и интеграция в умный дом хотят инструкции - вот вам умный дом, напишите такие-то команды и скрипты, нажмите на эти кнопочки. А такого нет.

А тот кто знает как построить связь MQTTЮ как это интегрировать в мный дом - им это (в этом проекте нафиг не нужно).

Поэтому - нет детального описания (технического задания) - нет и реализации. Так пока на этом и остаемся.

GeorgyBenelli, 2023-12-22:
> В прошлой версии задача MQTT выполняла на 90% обслуживание связи андроид-приложения для управления вне локальной сети. И все команды (формат команд) были заточены исключительно под это приложение. То как было реализовано там - это во многом противоречит нормальной, классической реализации связи по MQTT. Что вызывало многочисленные трудности и нарекания от тех, кто пытался построить на этом интеграцию с умным домом.
> 
> Приложения больше нет и в этом смысле MQTT перестал быть нужным - я убрал его.
> 
> То что вы говорите - это общие слова и пожелания без определения технической реализации. На этом код не напишешь и связь не построишь. Я понятия не имею как работают умные дома и как технически все это должно быть реализовано. Поэтому и не берусь за это дело.
> 
> И похоже что те, кому нужно управление по MQTT и интеграция в умный дом хотят инструкции - вот вам умный дом, напишите такие-то команды и скрипты, нажмите на эти кнопочки. А такого нет.
> 
> А тот кто знает как построить связь MQTT, как это интегрировать в мный дом - им это (в этом проекте нафиг не нужно).
> 
> Поэтому - нет детального описания (технического задания) - нет и реализации. Так пока на этом и остаемся.

Далеко ходить не буду. Например есть комьюнити Sprut по умным домам. Несколько тысяч активных юзеров из которых наверное минимум несколько сотен человек кто в состоянии собрать Ваш проект(паять есп или купить готовую Gledopto WLED и на нее накатить).Там официальные интеграции по апи выходят лишь на массовые продукты, а остальное делают обычные участники комьюнити и именно по мктт. Не знаю сколько из этих людей хотели бы повторить проект, но думаю когда узнают то будут такие же. 

Вы правы, это лишь мое пожелание. Я пользователь NodeRED и мне было бы проще автоматизации строить там на MQTT чем на апи, как и многим другим пользователям системы NodeRED

vortigont, 2023-12-23:
> Я пользователь NodeRED и мне было бы проще автоматизации строить там на MQTT чем на апи

нет такого понятия "строить автоматизацию на MQTT". Это всё равно что сказать "строить автоматизацию на HTTP". MQTT описывает лишь формат передачи данных, сырых данных. Т.е. это даже не текст. Что в этих данных будет бегать и как именно их интерпретировать (как текст, как бинарные данные, как джейсон определенной структуры) определяется строго формализованными договоренностями - это и есть API. Построить автоматизацию на MQTT не договорившись об API для устройства которое вы хотите автоматизировать - невозможно.
API в этом проекте есть, сделать так что бы те же текстовые сообщения которые бегают между вебмордой и контроллером начали бегать и в MQTT не сложно. Вопрос в том, готовы ли вы на стороне NodeRed написать всю ту логику, которая нужна для разбора и отправки этих сообщений что бы построить интеграцию?

vvip-68, 2023-12-23:
@GeorgyBenelli - ну вот, вы собираетесь строить автоматизацию на NodeRED. То есть вы знаете / понимаете что и в каком формате, используя какие каналы вы собираетесь отправлять в устройство и получать от него ответы. Ну вот и сформулируйте. Опишите весь протокол. Оформите в виде ТЗ.

Мне вот например всю эту автоматизацию, буде она реализована - даже тестировать не на чем.

GeorgyBenelli, 2023-12-23:
> @GeorgyBenelli - ну вот, вы собираетесь строить автоматизацию на NodeRED. То есть вы знаете / понимаете что и в каком формате, используя какие каналы вы собираетесь отправлять в устройство и [...]

MQTT в умнодомых системах не стандартизирован вообще.Каждый делает как хочет.Кто то делает один топик со всем в JSON,а кто то по топикам раскладывает(так удобнее).Обычно это топик состояния(state) и топик управления (command или set).И в системах умного дома нет никаких сложностей это обрабатывать, особых скриптов не нужно тк разработчики обычно заботятся о том чтобы это делалось легко.Достаточно указывать топики и что в них слать или брать с них.Есть конечно и немного стандартизированные проекты с MQTT такие как Home Assistant Discovery или [ESP Home](https://esphome.io/components/mqtt.html).Но в целом особо без разницы как оно будет.Вполне подойдет то что было в прошлой версии в MQTT;)

GeorgyBenelli, 2023-12-23:
[image]
Вот к примеру мой обычный аналоговый домофон.Все просто сделано и легко добавляется в любые системы.

closed 2023-12-24 by GeorgyBenelli