vvip-68/LedPanelWiFi #49: Мерцания, артефакты, паразитные пиксели

AlekseyMakshakov, 2023-12-23:
Добрый день! Сделал матрицу из ленты, 16*12, wemos d1 mini. На старой версии 1.13 все работает отлично. Поставил 1.14 - начались проблемы с мерцанием диодов, паразитные засветы вокруг зажигаемых диодов, лишние пиксели горят. Ставлю обратно 1.13 - опять все четко. Через веб-интерфейс пытаюсь менять яркость - меняются цвета и количество паразитных пикселей, яркость не меняется. Конденсатор, резистор, диод поставил - результат тот же, да и очевидно не в питании дело, раз на 1.13 все хорошо. Куда еще можно копать, не подскажете? 
Попробовал версию ядра 2.7.4 поставить (в 1.13 было написано, что  из-за этого подобное поведение может быть), но оно так даже не компилируется, ругается что какой-то pin неверно задан.
Версию ESP8266 пробовал как в документации написано - 3.1.2, FastLed 3.6.0

vvip-68, 2023-12-23:
Ну если версии ядра и библиотек соответствуют - думаю проблема все же в питании...
Ждите, может кто с таким сталкивался - ответит. 
У меня такой эффект только при плохом (шумящем БП) или завышенном напряжении питания замечался.

AlekseyMakshakov, 2023-12-23:
Спасибо за ответ! А разве может на разных прошивках эффект от питания настолько различаться?

GeorgyBenelli, 2023-12-23:
У меня тоже самое.И тоже собрал на пробу матрицу 16*12 на гирлянде.На 1.13 все хорошо.Залил  WLED и тоже все гуд.

vvip-68, 2023-12-24:
@AlekseyMakshakov ну, все-таки ядро микроконтроллера тоже большую роль играет. Да и сама библиотека FastLED. Помнится с пере[одом с 2.7.4 на 3.0.0 лента вообще перестала работать нормально. Всякие случайные вспышки, постоянное горение первого светодиода. Только к версии 3.1.2 более-менее стабилизировали. 

Синхронизирующего сигнала на этих лентах нет, все тайминги задаются программно. Возможно и лента немножечко "особенная" попалась и ядро накладывает свое влияние, сама библиотека.

@GeorgyBenelli и версия 1.13 и WLED - обе они используют ядро 2.7.4; Версия 1.14 использует ядро 3.1.2...

AlekseyMakshakov, 2023-12-24:
А 1.14 на 2.7.4 запустить не получится?)

vvip-68, 2023-12-24:
Нет. Там какие-то несовместимости в ядре уже, кажется. Впрочем - попробуйте, но думаю что нет.
Ну у 99% повторивших - проблем нет на ядре 3.1.2 и FastLED 3.6

ramballan, 2023-12-27:
у меня такая же проблема. Питание блока 5,5 в. Никаких конденсаторов и диодов не ставил, только резистор от ESP8266 к ленте. АртефЕсли отключить то вс ок работает

vvip-68, 2023-12-27:
Ну потому и мерцает, что без диода вместо допустимых 4.9В подано 5.5В

ramballan, 2023-12-27:
поставил диод, поведение не поменялось
попробовал другой блок питания (на нём 5.2)
попробовал подключить через понижающий dc dc и пробовал разные питания от 4-5 вольт
итог один и тоже, пока не подлючен wifi всё работает. как только подключил wifi начинаются артефакты

vvip-68, 2023-12-27:
Я не знаю как это у вас исправить. У меня нет идей.
Попробуйте взять прошивку WLED. Она, говорят, уже довольно функциональна и стабильно (как говорят) работает...

7918514, 2023-12-31:
> только резистор от ESP8266 к ленте.

Один раз у меня был случай, что именно из-за этого резистора были глюки. Как только резистор убрал, то сразу всё нормализовалось.