vvip-68/LedPanelWiFi #51: Контроллер зависает/тормозит на морозе?

saippuakauppias, 2023-12-29:
Заметил странное – на улице похолодало (а гирлянда и МК на улице) и эффекты с SD карты теперь зависают и фактически воспроизводятся меняя 1 кадр в несколько секунд.
После перезагрузки всё ок, но возможно только до следующего перехода на эффект с карты. 
Пока что эффекты с SD отключил понаблюдать, но может быть кто-то сталкивался с чем-то похожим?
Дебаг с серийного порта не делал, подключившись к МК, да и вряд ли там будет что-то. Но веб-интерфейс в этот момент точно не грузит.

PS: модуль для карты от амперки (http://wiki.amperka.ru/%D0%BF%D1%80%D0%BE%D0%B4%D1%83%D0%BA%D1%82%D1%8B:troyka-sd), какой был. Карта на 4ГБ памяти, всего где-то 30-40 эффектов по 2-5МБ (в сумме меньше нескольких сотен МБ).

vvip-68, 2023-12-30:
В ветке ntp_lwip выложил последнюю версию скетча, которая работает вполне сносно у меня сейчас. Попробуйте ее, если есть запасной контроллер, чтобы в случае чего просто поменять местами с текущим. Может оно не из за температуры зависает а из за ошибок в прошивке... Может с этой лучше будет работать

saippuakauppias, 2023-12-30:
Спасибо! Постараюсь попробовать просто прошить текущий МК, который стоит, а то запасного нету, да и текущий спаян вместе с SD-слотом и конденсатором, просто так не сменишь :)

К слову, сегодня заметил что такой же эффект с тормозами наблюдается и на бегущей строке (эффекты с SD пока отключил).

PS: Если буду подключать МК к ноутбуку в момент зависания, то что может быть в логах серийного порта? Может быть на что-то особенно стоит обратить внимание?

saippuakauppias, 2023-12-30:
PPS: к слову, больше недели всё работало без единого бага, но вот почему-то без всякого моего вмешательства (за исключением воздействия внешней среды в виде понижения температуры) – теперь он подвисает так(

vvip-68, 2023-12-30:
Похоже это зависит от фаз луны :(. Тоже то работало без замечаний, то фризы какие-то. Но я по большей части смотрю "трансляцию с мастера", а тут много зависит от того насколько забита сеть, смотрит ли кто-то на соседних компах кино с инета, к примеру. 

В логах монитора по этому поводу не будет ничего.
Не знаю, что у вас за контроллер, я сейчас на Wemos d1 mini гоняю...

Вот весь день сегодня перебирал версии FastLED и параметры компиляции. Пока лучшие результаты давал FastLED 3.4, c 
#define FASTLED_ALLOW_INTERRUPTS 0
ПЕРЕД
#include <FastLED.h> и примерно следующими настройками

[image]

vvip-68, 2023-12-30:
Flash mode можно QOUT ставить, но некоторые контроллеры с таким значением даже не запускаются. Тогда DOUT

saippuakauppias, 2023-12-30:
> В логах монитора по этому поводу не будет ничего.
Может быть стоит добавить логирование на карту памяти? Я бы погонял и скинул логи для изучения.
Например, каждый эффект/каждые 10 секунд логировать время + объем RAM/занятую CPU/что-то ещё, что косвенно может сказать о наличии проблеме где-то. Только главное дескриптор к файлу не держать открытым, что при перезагрузке МК – все точно flush-нулось в файл :)) Ну и за объемом файла, наверное, тоже следить и ротировать это как-то (как logrotate в *nix, например).

> Не знаю, что у вас за контроллер, я сейчас на Wemos d1 mini гоняю...
У меня ESP32, я сразу купил пожирнее, чтоб вывезло большую ленту (читал в вики это где-то)

По остальному – спасибо, буду пробовать) Уже только скорее всего 1-2 января, чтоб завтра случайно не убить МК перед новогодней ночью)))

vvip-68, 2023-12-31:
Логи. В данном случае ничего не дадут. Я гоняю "на прогоне" со включенным монитором порта и включенным отладочным выводом - вижу и остатки памяти и фрагментацию. Ничего особо не дает для понимания. 

Фро открытые файлы и пр. Насколько я помню LittleFS специально разработана таким образом, чтобы записанное сразу сохранялось и не портилось при внезапном пропадании питания / перезагрузке. Надежная файловая система.

Мудрое решение не трогать до НГ. Пусть так, но работает. А с новой версией - непонятно что. Пока выводы какие-то неутешительные...

С FastLED 3.4 заеданий текста практически нет - движется ровно, но...
С выключенными прерываниями - '#define FASTLED_ALLOW_INTERRUPTS 0' все выводится без задержек, но регулярно перезагружается с сообщением 'Exception 4: Level1Interrupt: Level-1 interrupt as indicated by set level-1 bits in the INTERRUPT register' где-то внутри ядра и периодически по WDT.

C '#define FASTLED_ALLOW_INTERRUPTS 1' - не падает, но постоянно горит первый диод.

На версиях 3.5/3.6 - текст дергается, диод первый то горит то не горит, падений меньше но есть, бывают замирания вывода на матрицу когда первый диод мерцает, остальные отображают предыдущую картинку.

Все это на ядре 3.1.2
Под стабильное ядро 2.7.4 - не собирается - говорит что программа не влазит в сегменты выделенной памяти.

Такая вот печальная история на данный момент.
Снова пытаюсь подобрать комбинации версий ядра и FastLED, опций компиляции...
Мне тут давно советовали бросить ESP8266 и уже писать только ориентируясь на ESP32

Там вроде более-менее все стабильно и без внезапных перезагрузок на ядре 2.0.14...

saippuakauppias, 2023-12-31:
> Мне тут давно советовали бросить ESP8266 и уже писать только ориентируясь на ESP32

Учитывая разницу в стоимости между ними в 150-200 рублей – как будто для конечного потребителя ничего не изменится в суммарной цене (лента всё равно сильно дороже МК).
Закинул донат на карту для покупки нескольких ESP32, авось и вправду можно на них отточить всё до идеала :)

vvip-68, 2023-12-31:
Спасибо за поддержку! :)

saippuakauppias, 2023-12-31:
Забыл дописать в предыдущем комментарии :)
Как временное решение – можно ещё добавить отдельную конфигурациионную настройку "через сколько эффектов подряд перезагружать МК", чтобы уменьшить количество сбоев

vvip-68, 2023-12-31:
Тут лучше по времени, скажем, в 4 утра.. Но бесполезно.
Вот вчера написал, что нашел лучший вариант - ядро 3.1.2 и FastLED 3.4 с запретом на прерывания.
Пока писал - вроде нормально все было. Потом перезагрузки начались каждые 10 минут Exception 4: Level1Interrupt: Level-1 interrupt as indicated by set level-1 bits in the INTERRUPT register' 

По итогу сейчас снова - FastLED 3.6, запрет на прерывания отсутствует. С утра крутится без падений. Текст бежит ровно. Может что еще в настройках компиляции поменял - не помню. Разве что частоту процессора со 160 вернул на 80.
А вчера на такой же конфигурации за исключением частоты - постоянные фризы эффектов и мерцание первого диода. Еще вчера роуутер дурил. Может от сети что-то зависит...

saippuakauppias, 2023-12-31:
А если от сети зависит, то может тогда выключать wi-fi спустя 5 минут после запуска (если не было подключения через веб-интерфейс)? Ну и само собой включать-выключать его когда идет запрос за погодой/временем (но возможно это делается прямо перед запуском эффекта и тут много доработок потребуется).

Сделать, например, тактовую кнопку, которая будет включать wi-fi на следующие 5 минут, ожидать подключения и потом снова отключать его. У меня тоже были мысли что это из-за сети может быть, но ни подтвердить, ни опровергнуть никак не могу :(

vvip-68, 2023-12-31:
Ну это извращение какое-то. То есть, когда я захочу подключиться к гирлянде что-то изменить, я не смогу этого сделать совсем или мне придется идти на балкон давить на кнопку? Да ну, бред какой-то...

Тут только наблюдением и вычислением совпадений. Чтобы выявить причинно-следственную связь.
И все равно у меня нет рецепта - сделать то-то и то-то. Там в меню "Инструменты" только параметров штук 5, которые могут влиять. А уж сколько комбинаций, да еще для различных версий библиотек. Плюс внешние непредсказуемые условия Тут как повезет вычислить наиболее стабильную комбинацию.

saippuakauppias, 2023-12-31:
Это да, итоговым решением назвать такое сложно, скорее как быстрый хак чтоб ещё сузить предполагаемые проблемы, на постоянку такое не очень, согласен

saippuakauppias, 2024-01-01:
Добрый вечер и с наступившим новым годом!)

Пока что не перепрошивал контроллер, а ограничился наблюдением. Вчера весь вечер погода была до -10 градусов (скорее всего в корпусе чуть теплее), а сегодня уже -20. Вчера ни разу за 9 часов ничего не зависло, а сегодня за уже 6 часов – 2 раза :)

Причем в этот раз зависает как-то интересно: эффекты меняются через какое-то странное время (установленно время смены эффекта 20 секунд), диоды все меняются одновременно и без задержек, а вот проход по кадрам внутри самого эффекта – тормозит. Само собой wi-fi отваливается в этот момент. Всё таки грешу на погодные условия и подумываю о том, чтоб поискать какой-нибудь обогревающиий элемент в корпус, который от 5В запитать можно.

Даже видео в этот раз записал: https://youtu.be/sxOXLpYAws0

vvip-68, 2024-01-02:
Так то по даташиту микроконтроллер должен работать от -40... Вот для ленты - нижняя заявленная граница -25....
У меня две зимы гирлянда на неотапливаемом балконе отвисела и в -30 работало...  Ну что ж - экспериментируйте....

closed 2024-01-09 by vvip-68