vvip-68/LedPanelWiFi #55: Установка на esp32-s2

didos06, 2024-01-13:
Наверное самая подходящая плата для этого проекта - esp32-s2 mini. Размер как у D1 mini, при этом цена не на много больше, в отличие от других версий esp32. Прошивка установилась, правда плагин little fs из инструкции не подходит, эта плата не поддерживается. Использовал  [это](https://github.com/lorol/arduino-esp32fs-plugin). 
Использовал Arduino ide1.8.14, ядро 2.0.14, поскольку Arduino ide2.x плагины вообще не поддерживает.
 Вроде всё работает. Правда у меня самая простая схема - матрица и обычная кнопка. Все эффекты работают, "дождь" тупит иногда. 
Важное дополнение по схеме - большинство проблем с артефактами решает установка резистора 2-10 кОм. между пином ленты и Gnd, даже самые проблемные ленты(светодиоды) отлично работают, тогда как ухищрения с конденсатором, уменьшением питания до 4,8В , диодом и последовательным резистором не помогали.
По web интерфейсу - при выборе эффекта из списка вручную - после перезагрузки он не запоминается пока не подвигаешь переключатель автосмены или случайного выбора. Если выбирать кнопками назад, вперёд то выбранный эффект сохраняется.

vvip-68, 2024-01-13:
ESP32-S2 уж точно не самая лучшая плата. По крайней мере на этапе ввода в эксплуатацию
Во первых она одноядерная в отличии от нормальной ESP32-d1-mini, к примеру.
Во вторых в ней памяти в два раза меньше по сравнению с (по крайней мере то что отображается в логе в мониторе порта).
В прошивку ее нужно вводить только вручную, нажимая кнопки. Из прошивки тоже выводить вручную, опять же нажимая кнопку сброса.
При прошивке она садится на один COM-порт, к примеру COM11, в режиме наблюдения при нормальной работе скетча - на COM12

Зато если в обычную ESP32 на 4M скетч уже не помещается при "все включено", в S2 еще остается немного свободного места. Из за этого на текущий момент в ESP32 приходится отключать поддержку ОТА.
Видимо придется выкидывать какие-то эффекты типа Анимация и Погода для освобождения памяти или брать платки на 8/16 мегабайт...

В ветке main 1.14 в эффектах traffic ошибка, которая на неквадратных матрицах может крашить систему. В нем же и еще в паре - не освобождается память. Эффект Дождь, кстати, довольно прожорлив по памяти...

Все это и исправление про незапоминающийся эффект уже лежит в ветке ntp_lwip, которая все еще в разработке, но по крайней мере эти ошибки там исправлены, проделана некоторая оптимизация, которая, надеюсь, приведет к более стабильной работе. Но, вполне вероятно,  добавлены и какие-то новые ошибки тоже...

didos06, 2024-01-13:
Да я по цене её сравнивал с Wemos D1 mini,   она самую малость дороже. С нормальной esp32(которая уже ощутимо дороже )всё правильно, она уступает количеством ядер, правда у неё оно посвежее.Да и чипы разные могут китайские друзья подсунуть, с меньшей памятью(правильный ESP32-S2FN4R2). Да и по портам прыгает - неудобство конечно, походу из-за встроенного  USB.

vvip-68, 2024-01-14:
Когда я писал про память - имелось в виду оперативная память. 520K против 320K. Память никогда лишней не бывает. Вон на Wemos ее всего 160K и уже как-то ощущается нехватка...

По цене. Примерно 180р у ESP32S2 против 350р у ESP32. Ну для школьника может и существенная разница (в два раза!!!) или при больших партиях закупать. А для среднестатистического DIY-щика - да он на пиво/сигареты больше тратит в разы...

didos06, 2024-01-14:
Поставил последнюю из ntp_lwip. В web интерфейсе несколько нижних эффектов местами перепутаны. Пока тестирую.

vvip-68, 2024-01-14:
Да, я знаю. Потому она и не в релизе. Думаю как более просто поставить их на место, чтобы без глобальных переделок.
Там еще кучка глюков, которые я периодически вылавливаю и исправляю. Сегодня вот поправил - строки в backup не сохранялись / не восстанавливались...

Хужее всего, что нестабильно работает трансляция в режиме группового вещания.  Как-то все с дерганием.
Вне этого режима вроде гладко, а  групповая синхронизация - просто увы. :(

Еще не могу подобрать нормально работающие библиотеки WebServer`a. Пробовал 4 разные, но какие-то просто в процессе работы "зависают" и web-сервер перестаёт отвечать, а иногда и пинговаться даже. Некоторые постоянно рвут связь со страничкой. Некоторые столько места занимают, что прошивка уже в память не влазит.

В общем - пока не очень стабильно. Всё в разработке. Но если есть желание потестить и найдете какие-то баги - welcome, сообщайте. И следите за обновлениями.

Если обновляется WebUI - необходимо сбрасывать кэш браузера или хотя бы обновлять страничку через Ctrl-Shift-R мли Ctrl-5, иначе файлы будут браться из кэша с диска, а не загружаться с устройства. Тогда страничка будет неправильно отображаться или даже вовсе не загрузится...

vvip-68, 2024-01-14:
@didos06 - исправил порядок эффектов. Если интересно - обновите, в том числе и файловую систему.
Перед обновлением если есть SD-карта - лучше сохранить на нее текущие настройки, потом восстановите после обновления. С самими настройками вроде ничего случиться не должно, а вот текст бегущей строки будет стёрт. Он хранится в файловой системе - а она заменяется на новую при обновлении.  Сохраненные в файловой системе настройки тоже будут потеряны.  А вот с SD-картой ничего не делается - на ней сохраненные настройки остаются нетронутыми.

didos06, 2024-01-14:
Обновил. Порядок поправился. Ещё на предыдущей заметил - при выключении ночных часов автосмена включается. И не пойму каким цветом дождь задумывался? Уменьшаешь контраст - он синий, прибавляешь - какой-то разноцветный. Причём это ещё и от общей яркости зависит.

vvip-68, 2024-01-15:
Ночные часы. Дда, включается автосмена. Потому что команда "Включить ночные часы" в API есть - '$14 3' - то есть включить "специальный режим", а вто команды выключить ночные часы - нет. Они выключаются отправкой команды API "включить демо-режим". Как-то так сложилось исторически. Ну, если это реально сильно мешает - подумаю над добавлением новой команды...

Про температуру - е понял. Как это воспроизвести?
Включаю ночные часы (кнопка в интерфейсе). Текущий эффект сменяется на отображение ночных цветов (у меня красный). Еще раз тыкаю на кнопку. Включается какой-то из режимов. Погода из красной в часах становится синей (сейчас минусовые температуры). Опишите последовательность действий, приводящих к тому, что температура оставалось бы красной. оставалась 

По Дождю. Эффект не мой, цельнотянутый давно откуда-то, уже не помню.
Насколько я понимаю задумку - в эффекте используются все оттенки голубого - от темных к синему / цвету морской волны, до светлых. Капли - больше к белому, но тоже из серии голубых оттенков.
Изменение контрастности - по сути то же изменение яркости  эффекта по отношению к яркости часов.
Вполне естественно, что уменьшение яркости делает светлые сине/зеленые оттенки более темными - то есть цвета уходят в область синих/зеленых. Неяркие компоненты цвета RGB - в частности К и относительно - G, в силу особенности строения светодиодов - вообще перестают светиться, оставляя на матрице синий рисунок.

Ну, это мое понимание как это работает. Но такое поведение цвета - оно общее для всех эффектов. Это принцип работы светодиода. С уменьшением яркости остаются только более яркие компоненты цвета. Красный вообще светится даже когда чисто-белый цвет с равным значением компонент R,G,B - G и B уже погасли, К продолжает светиться.
Диоды серии WS2812 вообще не лучшие в этом плане. У них ШИМ- регулировка яркости  достаточно низкочастотный.

vvip-68, 2024-01-15:
@didos06 - я добавил отдельную команду выключения ночных часов в API

Нужно обновить не только саму прошивку, но и файлы странички (загрузка файловой системы), т.к. при нажатии на кнопку "Выключить ночные часы" теперь будет отправляться другая команда.

Нужно очистить кэш браузера (или хотя бы обновить страничку Ctrl-Shift-R)? чтобы код странички загрузился из контроллера, а не из кэша браузера с диска компьютера / телефона

didos06, 2024-01-15:
С температурой такая хрень происходит только когда выключен случайный выбор. Сначала не понял, решил для чистоты эксперимента прошить всё тоже самое начисто. Запустил - нормально всё с температурой. И как только отключил случайный выбор всё повторилось.

7918514, 2024-01-15:
> Важное дополнение по схеме - большинство проблем с артефактами решает установка резистора 2-10 кОм. между пином ленты и Gnd, даже самые проблемные ленты(светодиоды) отлично работают, тогда как ухищрения с конденсатором, уменьшением питания до 4,8В , диодом и последовательным резистором не помогали.

То есть управляющий пин ленты подтянуть через резистор к земле? Интересненько. Спасибо за информацию, первый раз слышу о таком!

vvip-68, 2024-01-16:
@didos06 
>С температурой такая хрень происходит только когда выключен случайный выбор. Сначала не понял, решил для чистоты эксперимента прошить всё тоже самое начисто. Запустил - нормально всё с температурой. И как только отключил случайный выбор всё повторилось.

Все равно не могу воспроизвести ошибку. Как только отключаешь ночные часы - всё восстанавливается, все цвета в норме.

didos06, 2024-01-16:
Так на последней ошибка ушла. Была на предыдущей, при отключённом случайном выборе. Теперь ночные часы корректно отключаются.

didos06, 2024-01-16:
Ещё вопрос, предложение, хотелка. Матрица 32х16. Эффект часы - вверху часы, внизу температура. А можно чтобы бегущая строка вместо температуры снизу пробегала, а сами часы на месте оставались? Строка пробежала - температура на место вернулась. И на матрицы 16х16 тоже бы подошло.

vvip-68, 2024-01-17:
Можно подумать как это организовать... На 16x16 будет некрасиво смотреться, когда часы прилеплены кверху, температура к низу матрицы, между ними пустота. Иначе бегучка не влезет или часы "прыгать" вверх-вниз будут

closed 2024-01-24 by vvip-68