vvip-68/LedPanelWiFi #56: пропажа бегущей строки

bdu1977, 2024-01-13:
бегущие строки у меня есть как простые, так и со скриптами. автосмена эффектов стоит через 60 сек, показ бегущей строки через 45 сек. до нового года бегущие строки показывались все и простые и со скриптами. а недавно заметил, что бегущая строка не отображается. при нажатии в приложении кнопки просмотр, строка выводится, но дальнейшего показа нету. как вернуть показ бегущей строчки?

vvip-68, 2024-01-14:
Ну пока такая проблема только у вас наблюдается. Больше никто не заявлял. У меня проблем с бегущей строкой не наблюдается. Значит это конкретно вашу систему смотреть нужно.

Не выключился ли общий флажок на страничке "Тексты" - "Текст поверх эффектов"? 
Не выключился ли "Бегущая строка поверх эффекта" в настройках каждого конкретного эффекта?
Какая конкретно версия прошивки стоит?

Если версия из ветки разработчика, а не из "main" в репозитории - там поменялась карта распределения памяти хранения настроек - настройки эффектов (да и параметров в целом) нужно выполнить с нуля.

bdu1977, 2024-01-14:
прошивка 1,13 стоит из папки "main" 
"текст поверх эффектов" - галка стоит
 "Бегущая строка поверх эффекта" в настройках каждого конкретного эффекта - оставлено только на тех эффектах где хотел бы видеть "бегущую строку". может есть где-то настройка, что бегущая строка в приоритете - и тогда все равно какой включен эффект, если подошло время показать строку, то она включается, не зависимо от эффекта.
но опять же до НГ все работало, а тут пропало. 
попробую еще все настройки перенастроить с нуля, но это будет чуть позже.

vvip-68, 2024-01-14:
Ну, 1.13 уже давно никто не трогал. Не знаю что с ней могло случиться.