vvip-68/LedPanelWiFi #59: DFplayer mp3

Andriejus, 2024-02-11:
Hi. In order to get sound from speaker i have to press "Show" two times. Why mp3 player does not react when Im pressing this button only one time? Any ideas? 

[image]

vvip-68, 2024-02-11:
In fact you have an empty string. After processing sound macros and exclude them from text, you get quite empty string - nothing to display. In this sense you shouldn`t hear any sound at all. That fact that you hear something after second click on 'Show' button - definitely the bug. You must not hear anything in your case. Probably you hear sound by the transition process between one empty string stopping and next (second press) empty string that will not be shown.

What happening if you put into text something else except {A} macros?

vvip-68, 2024-02-11:
It is really bug with no plat sound on first click 'Show' / first time show text. :(
I found it. Thank you. Will be fixed in next version published in this project storage.

vvip-68, 2024-02-15:
I fixed this issue in version WiFiPanel v.1.14с.2024.0215. It is uploaded into GitHub storage. Should work now. Need to check.

closed 2024-02-15 by vvip-68

Andriejus, 2024-02-18:
> I fixed this issue in version WiFiPanel v.1.14с.2024.0215. It is uploaded into GitHub storage. Should work now. Need to check.

Thanks. It works now!