vvip-68/LedPanelWiFi #62: Часовой пояс

ferakond, 2024-06-09:
Контроллер сбрасывает настройки времени. Примерно раз в час, и при перезагрузке. Настраивал часовой пояс в веб интерфейсе. С прошлой версией такой проблемы не было.

vvip-68, 2024-10-25:
Долгое время не заглядывал в проект... Извините за долгий ответ. Может уже и не нужно, но все же...
Да, видимо с выбором зоны какой-то косяк в прошивке - посмотрю / поправлю.
Пока же - найдите в настройках параметр типа этого:

#define TZONE PSTR("253_<+07>-7")

и укажите там зону для Иркутска - 243_<+08>-8

Зоны для других городов - в папке /data/web/assets/ есть файл tz-rus.json.gz - внутри него json файл со списком городов и временнЫх зон. Найти свой город, соответствующую ему временнУю зону и прописать в скетче.

vvip-68, 2024-10-25:
В версии от 26.10.2024 исправлено

closed 2024-10-25 by vvip-68