vvip-68/LedPanelWiFi #65: Compilation error: 'LINEARBLEND_NOWRAP' was not declared in this scope

hhrhhr, 2024-07-29:
Делаю всё по инструкции, при компиляции столкнулся с ошибкой, указанной в названии.
Поиск по исходникам выдаёт использование макроса LINEARBLEND_NOWRAP только в файле \firmware\LedPanelWiFi_v1.14\effects2.ino, но нигде нет определения этого макроса.

hhrhhr, 2024-07-29:
> не та версия FastLED используется

[Из вики](https://github.com/vvip-68/LedPanelWiFi/wiki/Пошаговая-инструкция-подготовки-среды-и-сборка-проекта):
_A строке поиска наберите "FastLED" Выберите блок "FastLED by Daniel Garcia"
Выберите версию библиотеки **3.5.0**_

Из README.MD
_Данный проект рассчитан на работу с версиями ядра плат ESP8266 версии 3.1.2, ESP32 версии 1.0.6 и библиотеки FastLED версии **3.5.0**;_

Из FAQ:
_**О**: Это стандартная библиотека для FastLED для управления адресными светодиодами. Идите в менеджер библиотек и установите ее. Версию библиотеки выбирайте **3.5** или 3.4;_

vvip-68, 2024-07-29:
Возможно в readme упустили, не актуализировали...
Думаю стоит ориентироваться на это:
[image]

hhrhhr, 2024-07-29:
Да, с FastLED 3.6 сборка прошла. Однако в версиях ниже 3.6 макроса LINEARBLEND_NOWRAP не существует, то есть комментарии про 3.4-3.5 уже не актуальны (без правки кода effects2.ino).

vvip-68, 2024-07-29:
Да, не актуальны... Надо будет как-нибудь поправить...

closed 2024-07-29 by vvip-68