vvip-68/LedPanelWiFi #77: как с этим бороться

Den19081981, 2024-10-31:
In file included from C:\Users\deniscekanov09\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer

vvip-68, 2024-10-31:
Это замечательно!

Den19081981, 2024-10-31:
как это победить ошибка выскакивает

vvip-68, 2024-10-31:
В этом сообщении нет никакой ошибки.

Den19081981, 2024-10-31:
215 | void ESP8266WebServerTemplate<ServerType>::requestAuthentication(HTTPAuthMethod mode, const char* realm, const String& authFailMsg) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\deniscekanov09\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:349,
                 from C:\Users\deniscekanov09\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:27,
                 from D:\Users\deniscekanov09\Desktop\гирлянда на окно\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\a_def_hard.h:1,
                 from D:\Users\deniscekanov09\Desktop\гирлянда на окно\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\LedPanelWiFi_v1.14.ino:290:
C:\Users\deniscekanov09\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:215:6: note: 'void esp8266webserver::ESP8266WebServerTemplate<ServerType>::requestAuthentication(HTTPAuthMethod, const char*, const String&)' previously declared here
  215 | void ESP8266WebServerTemplate<ServerType>::requestAuthentication(HTTPAuthMethod mode, const char* realm, const String& authFailMsg) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from D:\Users\deniscekanov09\Desktop\гирлянда на окно\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\a_def_hard.h:1,
                 from D:\Users\deniscekanov09\Desktop\гирлянда на окно\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\LedPanelWiFi_v1.14.ino:290:
C:\Users\deniscekanov09\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:233:6: error: redefinition of 'void esp8266webserver::ESP8266WebServerTemplate<ServerType>::on(const Uri&, esp8266webserver::ESP8266WebServerTemplate<ServerType>::THandlerFunction)'
  233 | void ESP8266WebServerTemplate<ServerType>::on(const Uri &uri, ESP8266WebServerTemplate<ServerType>::THandlerFunction handler) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\deniscekanov09\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:349,
                 from C:\Users\deniscekanov09\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:27,
                 from D:\Users\deniscekanov09\Desktop\гирлянда на окно\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\a_def_hard.h:1,
                 from D:\Users\deniscekanov09\Desktop\гирлянда на окно\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\LedPanelWiFi_v1.14.ino:290:
C:\Users\deniscekanov09\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:233:6: note: 'void esp8266webserver::ESP8266WebServerTemplate<ServerType>::on(const Uri&, esp8266webserver::ESP8266WebServerTemplate<ServerType>::THandlerFunction)' previously declared here
  233 | void ESP8266WebServerTemplate<ServerType>::on(const Uri &uri, ESP8266WebServerTemplate<ServerType>::THandlerFunction handler) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from D:\Users\deniscekanov09\Desktop\гирлянда на окно\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\a_def_hard.h:1,
                 from D:\Users\deniscekanov09\Desktop\гирлянда на окно\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\LedPanelWiFi_v1.14.ino:290:
C:\Users\deniscekanov09\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:238:6: error: redefinition of 'void esp8266webserver::ESP8266WebServerTemplate<ServerType>::on(const Uri&, HTTPMethod, esp8266webserver::ESP8266WebServerTemplate<ServerType>::THandlerFunction)'
  238 | void ESP8266WebServerTemplate<ServerType>::on(const Uri &uri, HTTPMethod method, ESP8266WebServerTemplate<ServerType>::THandlerFunction fn) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\deniscekanov09\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:349,
                 from C:\Users\deniscekanov09\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:27,
                 from D:\Users\deniscekanov09\Desktop\гирлянда на окно\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\a_def_hard.h:1,
                 from D:\Users\deniscekanov09\Desktop\гирлянда на окно\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\LedPanelWiFi_v1.14.ino:290:
C:\Users\deniscekanov09\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:238:6: note: 'void esp8266webserver::ESP8266WebServerTemplate<ServerType>::on(const Uri&, HTTPMethod, esp8266webserver::ESP8266WebServerTemplate<ServerType>::THandlerFunction)' previously declared here
  238 | void ESP8266WebServerTemplate<ServerType>::on(const Uri &uri, HTTPMethod method, ESP8266WebServerTemplate<ServerType>::THandlerFunction fn) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
                 from D:\Users\deniscekanov09\Desktop\гирлянда на окно\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\LedPanelWiFi_v1.14.ino:290:
d:\Users\deniscekanov09\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  157 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

vvip-68, 2024-10-31:
Затрудняюсь сказать... Попробуйте все сделать заново - 
удалить ядро 3.1.2, удалить из аппки Documents все библиотеки, скопируйте библиотеки из проекта и переустановить снова ядро 3.1.2

Оно ругается на какие-то дубли и переопределения в файлах ядра - как-будто ядро где-то установлено в двух разных (?) папках и два экземпляра файлов конфликтуют друг с другом...

vvip-68, 2024-10-31:
Тогда не знаю...

Den19081981, 2024-10-31:
Завтра на втором компе Винду переустановлю, всё скачаю по новой, и попробую. Потом отпишусь

vvip-68, 2024-10-31:
Только делайте пожалуйста все по инструкции.

Den19081981, 2024-11-04:
Вроде всё получилось, только периодически мигает 1й светодиод, и в это момент всё зависает где-то на минуту. И ещё, не могу понять как зайти в веб интерфейс

vvip-68, 2024-11-05:
По первому светодиоду и замиранию - здась:
https://github.com/vvip-68/LedPanelWiFi/wiki/FAQ#в-собрал-всё-работает-но-в-начале-матрицы-постоянно-ярко-горит-один-светодиод-как-исправить

По Web-интерфейсу - если в настройках уже задан IP - ввести его в браузере. Если не задан и создается точка доступа - подключиться к этой точке доступа и ввести в адресной строке браузера IP адрес вашей точки доступа. Обычно, если ничего не меняли - это 192.168.4.1

Если не залили в файловую систему микроконтроллера файлы web-странички - тогда сюда
https://github.com/vvip-68/LedPanelWiFi/wiki/FAQ#в-проект-собран-скетч-загружен-в-плату-при-попытке-подключиться-к-устройству-в-браузере-отображается-сервер-не-активен

closed 2024-11-06 by vvip-68

Den19081981, 2024-11-06:
С 1м светодиодом разобрался, с WEB интерфейсом не получается. Хотя он мне нужен на 1 раз, чтобы бегущая строка была не по верх эффектов, а на чёрном фоне, может просто можно что-то в скетче поправить, что бы бегущая строка была на чёрном фоне?

vvip-68, 2024-11-06:
Можно в бегущей строке указать макрос цвета фона. Для черного - {B#000000}
А что не получается с веб-интерфейсом?

vvip-68, 2024-11-06:
Спрошу (в третий, кажется), раз - а вы странички веб-интерфейса в файловую систему загрузили? *см. два поста выше).
По строке - это находится примерно здесь:
[image]

Den19081981, 2024-11-06:
Как вы написали, но у меня при нажатии 3х кнопок на клавиатуре не чего не появляется, версия Ардуино иде последняя установлена

vvip-68, 2024-11-06:
Вообще ничего не появляется? Странно. Это системная функция IDE - там с полсотни предопределенных команд...
Если совсем ничего - ну может быть нужно IDE переустановить...

Если не появляется меню загрузки файловой системы - а вы плагин то установили?
https://github.com/vvip-68/LedPanelWiFi/wiki/Загрузка-данных-в-файловую-систему-МК

Ну, если с последней версией IDE ничего не выходит - может старую установить - IDE 1.8.19?

Den19081981, 2024-11-06:
Завтра ещё буду пробовать. Если что 1.8.19 попробую установить

Den19081981, 2024-11-07:
спасибо огромное, на версии 1.8.19 все заработало, и веб интерфейс тоже

vvip-68, 2024-11-07:
Вот, не все новое одинаково полезное... Не спешите перескакивать на непроверенное новое - проблем можно поиметь множество на пустом месте...