vvip-68/LedPanelWiFi #78: Разные типы лент

R0mjke, 2024-11-02:
В прошлом году купил китайскую адресную ленту, прошил ESP, всё супер, целый сезон гирлянда отработала шикарно. В этом году заказал ещё три ленты по этому же лоту на АлиЭкспресс, чтобы сделать размер побольше. И вот возник небольшой нюанс, визуально все 4 ленты идентичные, но вот светодиоды внутри подключены по разному. На одной ленте порядок светодиодов GRB, а на трёх других RGB, соответственно, когда собираю одну большую ленту из 4 сегментов, на одном сегменте всегда другой цвет. В скетче нашел место, где указывается один общий порядок цветов в ленте. Соответственно вопрос, может быть как-то можно сделать, например в режиме сборной матрицы, чтобы каждому сегменту задавался свой тип ленты и порядок цветов в ней ?

vvip-68, 2024-11-03:
Два варианта решений.
1. Распределить вывод на разные пины, каждому пину определить свой порядок цвета.  В интерфейсе это тут - "Настройки", "Оборудование":
[image]

2. Вариант второй. В custom.ino найти вот этот вот кусочек кода:
[image]

До и после вызова FastLED.show() вставить два цикла по диодам ленты с другим порядком цветов (который кусок меньше - для того и цикл, а общий порядок цветов настроить в интерфейсе где на картинке выше

К примеру (абстрактно) два куска по 50 диодов, общее количество - 100
Один (первый) кусок имеет настроенный в интерфейсе порядок цветов GRB, Другой, например - RGB
То есть - нужно поменять две компоненты цвета R и G местами.

До FastLED.show() вставляем цикл типа

```
    for (uint16_t idx = 50; idx < 99; idx++) {
      uint8_t t = leds[idx].r;
      leds[idx].r = leds[idx].g;
      leds[idx].g = t;
    }
```

после - меняем в обратную сторону, возвращая как было
```
    for (uint16_t idx = 50; idx < 99; idx++) {
      uint8_t t = leds[idx].r;
      leds[idx].r = leds[idx].g;
      leds[idx].g = t;
    }
```

Ну или для другого порядка - поиграться с компонентами r,g,b. Возможно не только к и п нужно менять, а по кругу типа
r->g, g->b, и b->к - зависит от конкретного порядка в ленте. После FastLED.show() - все вернуть на место.

Это общий принцип - нюансы смотрите на месте.

R0mjke, 2024-11-03:
Огромное спасибо! С помощью первого варианта удалось всё быстро исправить! Прошу прощения за невнимательность!

closed 2024-11-03 by R0mjke