vvip-68/LedPanelWiFi #80: Проблема с веб подключением

Del3456, 2024-11-12:
Загрузил в NodeMCU версию проекта 1.14, не могу в браузере открыть http://192.168.4.1, после того как  по WiFi подключаюсь к устройству. Оно не отвечает ни на пинг, не заходит в браузере на него. Браузер yandex. На скриншотах видно, что все вроде 
должно работать, куда копать?

Вчера ставил версию 1.13 с приложения все подключалось к плате корректно

[image]
[image]

vvip-68, 2024-11-12:
Попробуйте отключить в телефоне передачу мобильных данных, оставив только WiFi

vvip-68, 2024-11-12:
Ну да. Не залил файловую систему в микроконтроллер. Там все скрипты страницы находятся. 
Или не залилось - многие сообщfли, что под 2.3.3 все время все глючит и зальется или нет - лотерея.
Если с 10 раз не получается - поставьте все же нормальную Arduino IDE 1.8.19 и залейте ее-шными плагинами.

vvip-68, 2024-11-12:
Какой плагин для заливки стоит? Что выбрано в разделе Инструменты - Partition Scheme?
Что пишет  в логе при загрузке файловой системы - нет ли там ошибок?

Del3456, 2024-11-12:
Все получилось, лагодарю!

closed 2024-11-12 by Del3456

dshrpv, 2024-11-16:
Коллеги. Доброго дня, ночи, полуночи, утра. Тоже не получается поднять интерфейс ((
Я тот еще мамкин "программист", хз куда смотреть копать.
После компиляции скетча (версия 1.14) "белого" сообщения об успехе нет. Есть гора сообщений, но прошивка в итоге собирается и заливается, лампочки светятся, но либо не поднимается DHCP либо при настройке подключения и статического адреса (в скетче) не поднимается вэб-интерфейс (File not found  URL:/).

Подскажите пожалуйста.

In file included from C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_hard.h:630,
                 from C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\LedPanelWiFi_v1.14.ino:290:
C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_lang.h:39:6: warning: multi-character character constant [-Wmultichar]
   39 | #if (LANG == 'RUS')
      |      ^~~~
C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_lang.h:39:14: warning: multi-character character constant [-Wmultichar]
   39 | #if (LANG == 'RUS')
      |              ^~~~~
In file included from C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_hard.h:630,
                 from C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\LedPanelWiFi_v1.14.ino:290:
C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_lang.h:343:6: warning: multi-character character constant [-Wmultichar]
  343 | #if (LANG == 'ENG')
      |      ^~~~
C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_lang.h:343:14: warning: multi-character character constant [-Wmultichar]
  343 | #if (LANG == 'ENG')
      |              ^~~~~
C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_lang.h:638:6: warning: multi-character character constant [-Wmultichar]
  638 | #if (LANG == 'ESP')
      |      ^~~~
C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_lang.h:638:14: warning: multi-character character constant [-Wmultichar]
  638 | #if (LANG == 'ESP')
      |              ^~~~~
In file included from C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_hard.h:630,
                 from C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\timeProcessor.cpp:23:
C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_lang.h:39:6: warning: multi-character character constant [-Wmultichar]
   39 | #if (LANG == 'RUS')
      |      ^~~~
C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_lang.h:39:14: warning: multi-character character constant [-Wmultichar]
   39 | #if (LANG == 'RUS')
      |              ^~~~~
In file included from C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_hard.h:630,
                 from C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\timeProcessor.cpp:23:
C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_lang.h:343:6: warning: multi-character character constant [-Wmultichar]
  343 | #if (LANG == 'ENG')
      |      ^~~~
C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_lang.h:343:14: warning: multi-character character constant [-Wmultichar]
  343 | #if (LANG == 'ENG')
      |              ^~~~~
C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_lang.h:638:6: warning: multi-character character constant [-Wmultichar]
  638 | #if (LANG == 'ESP')
      |      ^~~~
C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_lang.h:638:14: warning: multi-character character constant [-Wmultichar]
  638 | #if (LANG == 'ESP')
      |              ^~~~~
In file included from C:\Users\Lenovo\Documents\Arduino\libraries\FastLED\src/FastLED.h:75,
                 from C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_hard.h:656,
                 from C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\timeProcessor.cpp:23:
C:\Users\Lenovo\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  157 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Lenovo\Documents\Arduino\libraries\FastLED\src/FastLED.h:75,
                 from C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\a_def_hard.h:656,
                 from C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\LedPanelWiFi_v1.14.ino:290:
C:\Users\Lenovo\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  157 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
. Variables and constants in RAM (global, static), used 42748 / 80192 bytes (53%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ DATA     1808     initialized variables
╠══ RODATA   8532     constants       
╚══ BSS      32408    zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR), used 60967 / 65536 bytes (93%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ ICACHE   32768    reserved space for flash instruction cache
╚══ IRAM     28199    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 632088 / 1048576 bytes (60%)
║   SEGMENT  BYTES    DESCRIPTION
╚══ IROM     632088   code in flash

dshrpv, 2024-11-16:
> Все получилось, благодарю!

Отлично!
А что не так было? Как излечилось?

vvip-68, 2024-11-17:
В приведенном выше логе ошибок нет. Более того - приведенная в конце лога таблица разбивки сегментов скомпилированного кода говорит об успешной сборке скетча.

File not found URL: / 
говорит о том, что вы не залили файловую систему с файлами веб-странички в микроконтроллер - там все скрипты страницы находятся. Или не залилось - многие сообщали, что под 2.3.3 все время все глючит и зальется или нет - лотерея.
Если с 10 раз не получается - поставьте все же нормальную Arduino IDE 1.8.19 и залейте ее-шными плагинами.

https://github.com/vvip-68/LedPanelWiFi/wiki/Загрузка-данных-в-файловую-систему-МК

dshrpv, 2024-11-17:
Большое спасибо за быстрый ответ!
Да действительно, я (в силу отсутствия знаний и навыков или невнимательности) не закончил этот шаг.

И снова в замешательстве ((
Скопировал содержимое архива arduino-esp8266littlefs-plugin-master.zip по ссылке в инструкции
Перезагрузил программу
Плагин не установился.
В инструкции упомянуто имя архива ESP8266LittleFS-2.6.0.zip
Нагуглил  https://github.com/earlephilhower/arduino-esp8266littlefs-plugin/releases
Скопировал содержимое архива в c:\Program Files (x86)\Arduino\Plugins\, перезагрузил программу.
Плагин установился.

Но дальше продвинуться не получается.
При попытке загрузки, программа выдаёт сообщение
  
[LittleFS] data    : C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\data
[LittleFS] size    : 2024
[LittleFS] page    : 256
[LittleFS] block   : 8192
/arduino-littlefs-upload-1.3.0.vsix
unable to open '/arduino-littlefs-upload-1.3.0.vsix.
error adding file!

LittleFS Create Failed!

В инструкции упомянут arduino-littlefs-upload-1.2.1.vsix, но последняя версия arduino-littlefs-upload-1.3.0.vsix и в логах попытка открыть arduino-littlefs-upload-1.3.0.vsix
Я где это только можно подсунул их поочереди.
Без результата.

[LittleFS] data    : C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\data
[LittleFS] size    : 2024
[LittleFS] page    : 256
[LittleFS] block   : 8192
/arduino-littlefs-upload-1.2.1.vsix
unable to open '/arduino-littlefs-upload-1.2.1.vsix.
error adding file!

LittleFS Create Failed!
 
или

[LittleFS] data    : C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\data
[LittleFS] size    : 2024
[LittleFS] page    : 256
[LittleFS] block   : 8192
/arduino-littlefs-upload-1.3.0.vsix
unable to open '/arduino-littlefs-upload-1.3.0.vsix.
error adding file!

LittleFS Create Failed!

dshrpv, 2024-11-17:
попробовал загрузить через 2.3.3.

LittleFS Filesystem Uploader v1.2.1 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: C:\Users\Lenovo\Documents\Arduino\LedPanelWiFi_v1.14
   Data Path: C:\Users\Lenovo\Documents\Arduino\LedPanelWiFi_v1.14\data
      Device: ESP8266 series

Тоже ошибка ((

Building LittleFS filesystem
Command Line: C:\Users\Lenovo\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec\mklittlefs.exe -c C:\Users\Lenovo\Documents\Arduino\LedPanelWiFi_v1.14\data -p 256 -b 8192 -s 2072576 C:\Users\Lenovo\AppData\Local\Temp\tmp-36824-AtYCpoQOnN7k-.littlefs.bin
/arduino-littlefs-upload-1.2.1.vsix
unable to open '/arduino-littlefs-upload-1.2.1.vsix.
error adding file!

ERROR:  Mklittlefs failed, error code: 1

vvip-68, 2024-11-17:
А в какой папке плагин для выгрузки то находится?

vvip-68, 2024-11-17:
Когда писал инструкцию - последний был 1.2.1. Новый действительно 1.3.0
Скачал, установил, проверил - всё прекрасно загружается с этим плагином...

vvip-68, 2024-11-17:
Думал может с ESP8266 что-то не так... Но нет же...
```
LittleFS Filesystem Uploader v1.3.0 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: D:\DIY\Arduino\Projects\LedPanelWiFi\firmware\LedPanelWiFi_v1.14
   Data Path: D:\DIY\Arduino\Projects\LedPanelWiFi\firmware\LedPanelWiFi_v1.14\data
      Device: ESP8266 series

Building LittleFS filesystem
Command Line: C:\Users\AlVa\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933\mklittlefs.exe -c D:\DIY\Arduino\Projects\LedPanelWiFi\firmware\LedPanelWiFi_v1.14\data -p 256 -b 8192 -s 2072576 C:\Users\AlVa\AppData\Local\Temp\tmp-7664-D0rg0IQPOYL4-.littlefs.bin
/web/460.165f828294cdeda0.js.gz
/web/480.4a5de38231dac2ee.js.gz
/web/482.07dea5e05efa425a.js.gz
/web/489.b79619e5d87749bb.js.gz
/web/508.e4cb8ea546c73c86.js.gz
/web/51.c616389f35bf44c5.js.gz
/web/512.2694c16f82183863.js.gz
/web/978.fc7e5516e4f4fe76.js.gz
/web/assets/images/esp32.svg.gz
/web/assets/images/esp32c3.svg.gz
/web/assets/images/esp32c3mini.png.gz
/web/assets/images/esp32s2.svg.gz
/web/assets/images/esp32s2mini.png.gz
/web/assets/images/esp32s3.svg.gz
/web/assets/images/esp32s3mini.png.gz
/web/assets/images/nodemcu.svg.gz
/web/assets/images/wemos.svg.gz
/web/assets/langs/eng.json.gz
/web/assets/langs/esp.json.gz
/web/assets/langs/lat.json.gz
/web/assets/langs/rus.json.gz
/web/assets/tz-eng.json.gz
/web/assets/tz-esp.json.gz
/web/assets/tz-lat.json.gz
/web/assets/tz-rus.json.gz
/web/common.f9a5cf812b595b9c.js.gz
/web/favicon.ico.gz
/web/index.html.gz
/web/main.2398e7406950ae66.js.gz
/web/MI.122e921fbc0a837a.woff2
/web/pf.51221d56f96de468.js.gz
/web/runtime.497d25667d44c1d8.js.gz
/web/styles.2b31b41ad3aedf84.css.gz
/web/wallet.72f50a6de45ab9c2.svg.gz

Uploading LittleFS filesystem
Command Line: C:\Users\AlVa\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe C:\Users\AlVa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\tools\upload.py --chip esp8266 --port COM12 --baud 921600 write_flash 2097152 C:\Users\AlVa\AppData\Local\Temp\tmp-7664-D0rg0IQPOYL4-.littlefs.bin
esptool.py v3.0
Serial port COM12
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 30:83:98:b1:37:5d
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 2072576 bytes to 624011...
Writing at 0x00200000... (2 %)
Writing at 0x00204000... (5 %)
Writing at 0x00208000... (7 %)
Writing at 0x0020c000... (10 %)
Writing at 0x00210000... (12 %)
Writing at 0x00214000... (15 %)
Writing at 0x00218000... (17 %)
Writing at 0x0021c000... (20 %)
Writing at 0x00220000... (23 %)
Writing at 0x00224000... (25 %)
Writing at 0x00228000... (28 %)
Writing at 0x0022c000... (30 %)
Writing at 0x00230000... (33 %)
Writing at 0x00234000... (35 %)
Writing at 0x00238000... (38 %)
Writing at 0x0023c000... (41 %)
Writing at 0x00240000... (43 %)
Writing at 0x00244000... (46 %)
Writing at 0x00248000... (48 %)
Writing at 0x0024c000... (51 %)
Writing at 0x00250000... (53 %)
Writing at 0x00254000... (56 %)
Writing at 0x00258000... (58 %)
Writing at 0x0025c000... (61 %)
Writing at 0x00260000... (64 %)
Writing at 0x00264000... (66 %)
Writing at 0x00268000... (69 %)
Writing at 0x0026c000... (71 %)
Writing at 0x00270000... (74 %)
Writing at 0x00274000... (76 %)
Writing at 0x00278000... (79 %)
Writing at 0x0027c000... (82 %)
Writing at 0x00280000... (84 %)
Writing at 0x00284000... (87 %)
Writing at 0x00288000... (89 %)
Writing at 0x0028c000... (92 %)
Writing at 0x00290000... (94 %)
Writing at 0x00294000... (97 %)
Writing at 0x00298000... (100 %)
Wrote 2072576 bytes (624011 compressed) at 0x00200000 in 17.6 seconds (effective 940.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Completed upload.
```

dshrpv, 2024-11-17:
Cпасибо!

Плагины распихал везде, чтобы только накормить программу
c:\Users\Lenovo\OneDrive\Документы\Arduino\tools\
c:\Users\Lenovo\.arduinoIDE\plugins\
c:\Program Files (x86)\Arduino\Plugins\
c:\Program Files (x86)\Arduino\tools\

в 1.8.19
для загрузки нажимаю
[image]

в 2.3.3
для загрузки нажимаю (как в инструкции)
[Ctrl] + [Shift] + [P], затем выберите в появившемся меню "Upload LittleFS to Pico/ESP8266/Esp32".

Во всех случаях
unable to open '/arduino-littlefs-upload-1.2.1.vsix.
error adding file!

Попробовал на другом ПК, та же беда ((

vvip-68, 2024-11-17:
А ведь так и не написали, куда плагин arduino-littlefs-upload-1.2.1.vsix положили....
Может в этом то и дело? Хотя, тогда б, наверное, и в меню бы не появилось...
Ну, тогда не знаю. Ждите - может кто-нибудь ответит.

vvip-68, 2024-11-17:
А вы вообще читали КУДА его нужно положить?
Вот, к примеру, в проекте - ва папке tools/LittleFS_Uploader/Arduino IDE 2.3.x файл readme
Туда положили? И из всех прочих мест его лучше удалить. А то может в конфликт вступают...

dshrpv, 2024-11-17:
Да.
и тут лежит
[image]

vvip-68, 2024-11-17:
Тогда у меня нет ответа...

dshrpv, 2024-11-17:
сейчас удалю его отовсюду кроме этой локации
и попробую

dshrpv, 2024-11-17:
безуспешно ((

[LittleFS] data    : C:\Temp\Panel WS2812B WiFi\LedPanelWiFi-main\LedPanelWiFi_v1.14\data
[LittleFS] size    : 2024
[LittleFS] page    : 256
[LittleFS] block   : 8192
/arduino-littlefs-upload-1.2.1.vsix
unable to open '/arduino-littlefs-upload-1.2.1.vsix.
error adding file!

LittleFS Create Failed!

vvip-68, 2024-11-17:
Ну, у меня кончились идеи. Мне вот непонятно - я выше приводил логи - у меня вообще нигде не фигурировал путь C:\Temp\xxx...  У вас то почему он есть? Вы проект в папке C:\Temp\Panel WS2812B WiFi\ что ли держите?

dshrpv, 2024-11-17:
да, в этой папке с:\temp
а какая папка нормальная?

vvip-68, 2024-11-17:
Любая, которая не является по умолчанию системной папкой для временных файлов.
Впрочем, не знаю, что уже думать...

dshrpv, 2024-11-17:
Рад бы предположить, если бы разбирался в этом.

dshrpv, 2024-11-17:
Переместил каталог со скетчем в корень, тот же результат.
В папку C:\1\ , тот же результат
т.е. дело не в том где скетч ((

dshrpv, 2024-11-17:
А могули попросить создать уже готовый файл для заливки в node MCU с минимальным набором опций (кнопка D4, лента D2)?

vvip-68, 2024-11-17:
Я уже в исходный код плагина залез - там даже строк таких нет, которые бы это выводили в лог...
Вы точно этим плагином для выгрузки пользуетесь? Не нет там такого кода, чтоб так выводил.
Удалите вообще все плагины выгрузки. Идите в папку проекта tools\LittleFS_Uploader\Arduino IDE 2.3.x
возьмите плагин оттуда. В точности сделайте как написано в файле readme...

vvip-68, 2024-11-17:
Готовый файл все равно только прошивку грузит. А файловая система заливается отдельно вот этим вот плагином.

dshrpv, 2024-11-17:
> Я уже в исходный код плагина залез - там даже строк таких нет, которые бы это выводили в лог... Вы точно этим плагином для выгрузки пользуетесь? Не нет там такого кода, чтоб так выводил. Удалите [...]

при этом если я плагин кладу 1.2.1 или 1.3.0, программа их различает и пишет что не может открыть именно 1.2.1 или 1.3.0

vvip-68, 2024-11-17:
Без разницы какой. Они оба нормально работают. Но я бы настоял на том, что в комплекте с пректом идет.

dshrpv, 2024-11-17:
Начну всё с 0 (в очередной раз)

dshrpv, 2024-11-17:
Спасибо за поддержку и советы!
Надеюсь вернуться с хорошими новостями )))

vvip-68, 2024-11-18:
Поздравляю! 
Ну и в чем была проблема в предыдущих попытках?

dshrpv, 2024-11-18:
Т.к. все переделал с 0 (всё что можно снёс/удалил/переустановил) могу только предположить, что проблема была в arduino-esp8266littlefs-plugin со страницы разработчика.
В этой попытке использовал всё только из архива с проектом.
Возможно стоит убрать из инструкции ссылки на оригиналы плагинов, т.к. это создаёт проблемы/путаницу в версиях и их работоспособности.

Еще раз благодарю!!!!
Как за сам проект и его развитие.
Так и за оперативную поддержку/помощь.

hhrhhr, 2024-11-19:
> Думал может с ESP8266 что-то не так... Но нет же...

Плата LOLIN(WEMOS) D1 mini Pro, конфиг в плагине, соответственно, такой:
```
    ( 'd1_mini_pro', {
        'name': 'LOLIN(WEMOS) D1 mini Pro',
        'opts': {
            '.build.board': 'ESP8266_WEMOS_D1MINIPRO',
            '.build.variant': 'd1_mini',
            },
        'macro': [
            'resetmethod_nodemcu',
            'flashmode_dio',
            'flashfreq_40',
            '16M',
            ],
        'serial': '921',
        'desc': [ 'Product page: https://www.wemos.cc/' ],
    }),
```
То есть флешка 16М. При попытке заливки файлов веб-интерфейса всё почти как у вас, кроме пары ключей с размерами. У вас в логе `-s 2072576`, у меня `-s 14655488`. Подготовка файла проходит, создаётся временный файлик размером 14655488 байте, но при его загрузке облом (*** - вырезано для краткости):
```
Uploading LittleFS filesystem
Command Line: ***\python3.exe ***\upload.py --chip esp8266 --port COM5 --baud 921600 write_flash 2097152 ***.littlefs.bin
esptool.py v3.0
Serial port COM5
Connecting....
Chip is ESP8266EX
Features: WiFi
***
Configuring flash size...
Auto-detected Flash size: 4MB

A fatal esptool.py error occurred: File ***.littlefs.bin (length 14655488) at offset 2097152 will not fit in 4194304 bytes of flash. Use --flash-size argument, or change flashing address.
```

Как вручную указать размер флеш-памяти, чтобы образ создавался с учётом параметров платы, а не "_Auto-detected Flash size: 4MB_"?

hhrhhr, 2024-11-19:
> нужно выбрать разбивку партишнов выбрать 16MB(FS14MB/OTA~1019КВ)

Так и есть:
[image]

[Плата такая](https://sl.aliexpress.ru/p?key=jOY8roz).

Если ручками запустить обе команды (для подготовки образа и его заливки) с указанием размера не 14655488, а 2097152, то всё вроде бы замечательно:
```
***
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1048576 bytes to 621988...
Wrote 1048576 bytes (621988 compressed) at 0x00200000 in 14.9 seconds (effective 564.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
Но после перезапуска в логе видно, что каталог `/web` не найден, свободного места все оставшиеся 14 Мб.
Качаю "старую" IDE, может с ней что-то более осмысленное будет.

vvip-68, 2024-11-19:
А с чего вы решили, что она Pro, кроме того, что китаец так написал? По виду ничем не отличается от обычной 4MB карточки... Может не зря AutoDetect обнаруживаете в ней 4MB памяти то? На страничке продукта про 16МB тоже ничего не написано...

hhrhhr, 2024-11-19:
С того, например, что файл прошивки размером **больше 4 Мб** успешно загружается, плата стартует (при ручном указании IP адреса), какие-то эффекты по RGB-ленте бегают. Веб-сервер запускается и сообщает про "Сервер не активен".
Как ещё я могу проверить объём флешки?

vvip-68, 2024-11-19:
А что за прошивка у вас больше 4МБ?

hhrhhr, 2024-11-19:
Точно, прошивка около 700 Кб. Совсем глаз замылился после кучи экспериментов. 
В общем, уже поставил старую IDE, плату выбрал **D1 mini clone**, собрал, прошил, загрузил веб-интерфейс - вуаля, все работает. Памяти действительно 4 Мб получается.
Влеплю китайцу минус за мои мучения.