vvip-68/LedPanelWiFi #81: Нет получения адреса по DHCP

hhrhhr, 2024-11-19:
Выставил корректные имя и пароль точки доступа, нули в адресе:
```
#define NETWORK_SSID PSTR("***")
#define NETWORK_PASS PSTR("***")
#define DEFAULT_IP { 0, 0, 0, 0 }
#define GTW 1
```

В логе serial-монитора вижу такое:
```
Подключение к *** -> DCHP 
Нажмите на кнопку для прерывания подключения
и переход в режим точки доступа
......
WiFi подключен. IP адрес: (IP unset)
MDNS - запущен

HTTP server started
Свободно памяти: FM: 29832  Max: 29832  Frag: 0
Режим: Поток
Режим: Спектрум
```

Эффекты меняются, но IP-адрес не получен. В логе маршрутизатора есть только сообщение "`STA(ec:64:c9:**:**:**) set key done in WPA2/WPA2PSK.`" (это MAC-адрес платы), которое повторяется каждые 3-4 минуты.

Есть какие-то подводные камны с получением адреса через DHCP? Жёсткая привязка к локальному диапазону 192.168.0.0/24, например?

vvip-68, 2024-11-19:
Вот понятия не имею. Никогда мне этого не нужно было. Я и не проверял и не углублялся.
В документации на ESP-ire написано - для получения адреса по DHCP укажите адрес 0.0.0.0, что прошивка честно и делает.
А уж почему не получает - мне это даже как-то не интересно было, я и не проверял. 
Если вам нужно - попробуйте, поизучайте вопрос, поэкспериментируйте.

Потом и нам расскажете что было не так и почему. И заодно как при получении адреса по DHCP в каждый раз будете соединяться с ней из веб-интерфейса? Перебирать адреса искать на какой адрес посадил роутер панельку? Каждый раз подключаться к монитору порта смотреть на какой адрес на этот раз селе? Каждый раз открывать админку роутера смотреть какой на этот раз адрес получился?

А если прописывать в роутере жесткую привязку, чтобы он каждый раз выдавал устройству один и тот же адрес - чем это проще один раз задать статический адрес самому устройству в скетче или через веб-страничку? Веб-интерфейс даже, кстати, не поддерживает чтобы можно было ввести адрес 0.0.0.0.

В общем - я смысла в DHCP не вижу никакого и наверное ковыряться с этим даже не буду. Будем считать, что прошивка не поддерживает назначение адресов по DHCP.

vvip-68, 2024-11-19:
Можно конечно, по имени обращаться  - типа http://device.local - но то -- такое себе...

hhrhhr, 2024-11-19:
У меня в домашней сети одной подсети уже не хватает для всех устройств. Запоминать и на бумажке записывать какой IP у какого MAC-адреса — вот это и есть "такое себе".
Абсолютно все домашние устройства имеют привязку по имени, практически любой маршрутизатор это умеет. Вообще голова не болит, кто какой адрес получил, доступ по girlyanda-4.home или printer-panasonic.home, nas.home или camera-parkovka.home.
Для особо трудных случаев есть выделенный диапазон адресов, которые имеют привязку IP к MAC-адресу, но при этом всё равно получают нормальное человеческое имя в домашнем домене. Это действительно проще, настраивать всё в одном месте на маршрутизаторе.

Но это лирика, факт же в том, что получение динамического адреса не работает. "смысла не вижу", конечно же, аргумент мощный, но ведь работало оно раньше.

vvip-68, 2024-11-19:
Ну так этому коду, который за соединение с сетью отвечает - уже лет 5 и никто его не трогал.
Может от версии ядра зависит. Не знаю. Если разберетесь - сообщите что нужно сделать, чтобы у всех работало. Мне это точно не нужно и времени копаться в этом сейчас нет абсолютно. Когда будет свободное время - может поразбираюсь. Для меня это точно не приоритет номер 1. по описанным причинам.

У вас дома больше 250 устройств, что вся подсетка забита?

hhrhhr, 2024-11-20:
> У вас дома больше 250 устройств, что вся подсетка забита?

Одновременно нет, конечно же, но только всяких беспроводных поделок и виртуальных машин для их отладки около 150. К тому же есть 3 отдельных сегмента со своей адресацией, кому-то можно в инет, кому-то нельзя, некоторые устройства исклюительно друг с другом должны общаться. Всякие Tuya-девайсы умеют по blutooth работать, но обязательно требуют wifi для запуска.

> Если разберетесь

Точно не буду, логику программы я ещё могу поправить, но лезть в сетевые дебри - ну его, вас подожду ;)

hhrhhr, 2024-11-21:
```
@@ -1095,7 +1095,7 @@
       DEBUG('.');
       DEBUG(IP_STA[3]);
     } else {
-      WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE, INADDR_NONE);
+      //WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE, INADDR_NONE);
       DEBUG(F(" -> DCHP "));
     }

@@ -1136,8 +1136,13 @@
         if (wifi_connected) {
           // Подключение установлено
           DEBUGLN();
-          DEBUG(F("WiFi подключен. IP адрес: "));
-          DEBUGLN(WiFi.localIP());
+          DEBUG(F("WiFi подключен. IP: "));
+          DEBUG(WiFi.localIP());
+          DEBUG(F(", MAC: "));
+          DEBUG(WiFi.macAddress());
+          DEBUG(F(", hostname: "));
+          DEBUGLN(WiFi.getHostname());
+
           if (mdns.begin(host_name)) {
             DEBUGLN(F("MDNS - запущен"));
           } else {

```
Вот такое придумалось. В случае динамического адреса вызывать _WiFi.config(INADDR_NONE..._ не нужно. Вторая часть патча просто для вывода MAC-а и заданного имени хоста, а то я уже в трех платах запутался где какая ;)

На клоне D1 mini все заработало:
[image]

На ESP32 проверить не на чем, но как минимум для получения MAC-адреса там совершенно другая процедура.

vvip-68, 2024-11-21:
Ну, ок, с выводом MAC - посмотрю что на ESP32...
А что - вот закомментаривание той строчки позволяет нормально подключаться по DHCP?
То есть проблема решена?

Интересно почему во всех форумах пишут именно эту строчку длф конфигурирования подключения по DCHP?
У гих то как-то работает. Второй вопрос - насколько я помню, конфигурирование параметров сети вот  этим вот
WiFi.config() сохраняет сконфигурированное подключение сети где-то во флэш-памяти микроконтроллера.
И, к примеру, если оно раньше было сконфигурировано на статический адрес, почему считается, что бех вызова этого 
WiFi.config() с пустыми параметрами он будет использовать подключение по DHCP? а не ранее сконфигурированные и сохраненные параметры?

hhrhhr, 2024-11-21:
Я всё что нашёл - это то, что использование _WiFi.mode(WIFI_STA);_ сбрасывает запомненные настройки, а _WiFi.config(...)_ используют исключительно для конфигурации статического адреса. Ну ещё требование использовать setHostname перед mode и scanNetworks (при использовании такой фичи), ибо сканирование сбрасывает и имя и сетевые настройки.

С таким патчем после запуска платы через веб-интерфейс изменил динамический адрес на статику, перезагрузился, зашёл по новому адресу и вернул обратно 0.0.0.0 — работает, новый адрес получает.

Если честно, то я даже просто API для модуля ESP8266WiFi не нашёл, только куцые примеры, из которых совершенно непонятно, почему нужно делать так, а не иначе.

closed 2024-11-22 by vvip-68