vvip-68/LedPanelWiFi #82: Использование SD-карты, как?

hhrhhr, 2024-11-19:
Собрал проект с поддержкой SD-шилда:
```
#define USE_SD 1
#define FS_AS_SD 0
```
Спаял бутерброд из основной платы и считывателя, прошил, карточку на 2 Гб отформатировал в FAT32, скопировал на неё каталог \web из ***\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\data\web — веб-интерфейс не завёлся. Скопировал его через LittleFS, всё нормально, за исключением отсутствия видимости SD-карты в логе загрузки.

Как именно нужно/можно использовать карточку? Как форматировать, какая структура каталогов? В вики дельных советов не обнаружил. Сделал штору 30х20 из шести 10 метровых RGB-гирлянд по 100 светодиодов, встроенные эффекты проигрываются замечательно, бегущие строки и прочие часы отображаются без проблем. Но вот как именно использовать SD-карту — пока непонятно.
Нужно ли её вытаскивать и копировать веб-интерфейс плюс всякие ролики на компьюетере/телефоне или можно загружать всякое на карту прямо через веб-интерфейс самой гирлянды? Сохранение настроек на карту проходит без проблем, но на самой карте ничего не находится.

vvip-68, 2024-11-20:
Web-интерфейс никакого отношения к SD-карте не имеет. Все его файлы хранятся внутри микроконтроллера, в его файловой системе. Данные в файловую систему микроконтроллера могут помещаться только специальным плагином-загрузчиком.

SD карта используется только для резервного копирования настроек прошивки (или в файловую систему или на SD-карту), сохранения рисунков пользователя, которые он натыкал на экране в режиме "Рисование" (или в файловую систему или на SD-карту по выбору) - рисунки сохраняются в папку формата WpH - где W- ширина матрицы, H - высота. В вашем случае - '30p20'. Далее сохраненные рисунки )или подготовленные в редакторе JinxFramer будут использованы для эффекта "Слайды".

Также на карточке хранятся ролики, подготовленные из Jinx для эффекта "SD-карта". Ролики должны размешаться в папке формата WxH, где W -ширина матрицы, H - высота матрицы, в вашем случае - '30x20'. Как подготовить ролики и скопировать их на SD-карточку подробно описано в материале Wiki проекта

https://github.com/vvip-68/LedPanelWiFi/wiki/Экспорт-эффектов-анимации-из-ПО-«Jinx!»

То, что вы можете сохранять на карту резервную копию настроек на SD-карту, говорит о том, что карта нормально подключена и распознается микроконтроллером. Осталось только наделать роликов в Jinx (как - описано по ссылке выше) и наблюдать их в эффекте "SD-карта"

vvip-68, 2024-11-20:
Сразу после нажатия на кнопку reset на карточке контроллера в локе должно быть следующее:
В блоке "Доступные возможности"
[image]

'+' в выделенной строке говорит о том, что в прошивке работа с SD-картой включена
Далее, ниже

[image]

говорит о том, что микроконтроллер нашел подключенную карточку
Далее проверяется наличие в корне файловой системы карты папки со слайдами (в моем случае - 35p10)
и выводится список найденных слайдов или сообщение о том, что слайдов нет.

[image]

Еще ниже - поиск и обнаружение роликов для эффекта "SD-карта" в папке с роликами (в моем случае - 35х10)

[image]

hhrhhr, 2024-11-20:
Разобрался, спасибо.
Карточку просто в телефоне отформатировал как съёмный накопитель, получилась FAT32. До этого Win10 вроде бы тоже в FAT32 форматировала, но, может быть, размер сектора какой-то другой выставила, из-за чего карта упорно была "не подключена".

closed 2024-11-20 by hhrhhr