vvip-68/LedPanelWiFi #89: осталось до Нового года

NLObP, 2024-12-05:
Не правильно считает оставшиеся дни. Сегодня (5 декабря), проверьте: до 13 декабря осталось 7 дней, а до 14 декабря осталось 9 дней.

vvip-68, 2024-12-06:
Сегодня 6 декабря проверил остаток до 14 и 15 декабря (сдвинул на день, чтобы соответствовать условиям)
Пишет что до 14 декабря осталось 7 дней 14 часов (ну или если округлять по правилам гугля - 8 дней), а до 15 декабря, опять же по правилам гугля - 9 дней. Что не так?

NLObP, 2024-12-06:
Нет, тут именно 13 и 14 декабря.

NLObP, 2024-12-06:
Начал искать причину из-за того что на одном окне (где старая прошивка), показывает до Нового года правильно, а на  другом окне с новой прошивкой на один день больше.

vvip-68, 2024-12-06:
Потому что в старой прошивке вычисление проводилось математически - сколько дней осталось, столько именно и показывает (отбрасывая часы). На что мне пользователи многократно тыкали носом, что Гугль показывает по другому - там даже если осталось 7 дней 2 часа - все равно показывает 8. То есть округляет в большую сторону. То есть у меня ошибка вычисления. Ну ок, сделал так, как гугль считает.

Теперь другая половина говорит, что у меня ошибка...

NLObP, 2024-12-06:
И как мне быть? Я на старой не хотел обновлять прошивку.

vvip-68, 2024-12-06:
Ну, видимо, так и оставаться на старой прошивке

NLObP, 2024-12-06:
А в новой сложно поменять? Если вы мне скажете что поменять?

vvip-68, 2024-12-06:
Ну, ч думаю это где-то в runningText.ino примерно тут:

[image]

NLObP, 2024-12-06:
Спасибо