vvip-68/LedPanelWiFi #90: При входе на web падает подключение

barkas803, 2024-12-10:
Добрый день уважаемый разработчик :)
Подскажи пожалуйста, в чем может быть дело, использую свежую версию LedPanelWiFi_v1.14, прошиваю wemos d1 mini, возникает проблема при входе на web-морду wifi-панели, web-морда загружается но в хаотичном порядке выкидывает "Соединение утеряно" подключаюсь через web-морду потому, что нет техники на android.
Происходит как в режиме Точка доступа так и при подключении к wifi - успешно работает но как только зашел на web-морду начинаются приколы, так же спустя пару минут нахождения в web-морде wemos рестартится.

Приложил лог и скрин, прошу помощи, при компиляции ошибки только про локализацию языка, все библиотеки ставил согласно скетчу, пробовал Arduino 1.8.19 и Arduino 2.3.4

Если в web-морду не лезть то всё работает судя по монитору порта.
Гирлянду еще не подключал, но пробовал подключать 5V от внешнего блока питания - результата нет, всё так же.

Помоги пожалуйста добрый человек :)

`15:51:25.464 -> FM: 18656  Max: 15408  Frag: 17
15:51:25.702 -> WebSocket клиент #10 отключен
15:51:26.144 -> WebSocket клиент #11 с адреса 192.168.4.2
15:51:27.163 -> WEB пакeт размером 20
15:51:27.163 -> <-- $6 7|NW|NX|IP|IPC|DH
15:51:27.163 -> FM: 19800  Max: 14104  Frag: 27
15:51:27.163 -> WEB пакeт размером 10
15:51:27.163 -> <-- $6 7|WZ|E0
15:51:27.163 -> FM: 19024  Max: 14104  Frag: 24
15:51:27.163 -> WEB пакeт размером 67
15:51:27.163 -> <-- $6 7|W|H|HN|CRLE|PS|LG|VR|FS|SX|CH|CV|WZ|AL|SM|UP|FM|PN|MC|MX|MZ|TM
15:51:27.163 -> FM: 18736  Max: 14104  Frag: 23
15:51:27.265 -> WEB пакeт размером 152
15:51:27.265 -> <-- $6 7|FM|UP|W|H|HN|CRLE|PS|LG|VR|FS|SX|CH|CV|WZ|AL|SM|PN|MC|MX|MZ|TM|LE|E0|M10|M11|M0|M1|M2|M3|M4|M5|M6|M7|M8|M9|SZ|PZ0|PZ1|2309|PW|EE|BF|IPC|DH|AN|AB|AU
15:51:27.302 -> FM: 18320  Max: 14104  Frag: 21
15:51:27.639 -> WEB пакeт размером 7
15:51:27.639 -> <-- $6 7|LE
15:51:27.639 -> FM: 19896  Max: 14104  Frag: 27
15:51:27.676 -> WebSocket клиент #11 отключен`

[image]

vvip-68, 2024-12-10:
А сколько диодов то в гирлянде. Судя по остаткам памяти - ее остается довольно мало. Хотя, вроде, и не критически мало. Вероятно библиотеке вебсервера для работы не хватает памяти и прошивка падает...

vvip-68, 2024-12-10:
Ну тогда не знаю. У меня на конфигурации из двух матриц 16х16 - то есть всего 512 диодов - работает без проблем. Так что даже не подскажу...

barkas803, 2024-12-10:
> Ну тогда не знаю. У меня на конфигурации из двух матриц 16х16 - то есть всего 512 диодов - работает без проблем. Так что даже не подскажу...

А может прошить не как wemos ?

barkas803, 2024-12-10:
> Ну тогда не знаю. У меня на конфигурации из двух матриц 16х16 - то есть всего 512 диодов - работает без проблем. Так что даже не подскажу...

может wemos с обманом :) брал на валберис :/

vvip-68, 2024-12-10:
Сколько их закупал - ни разу не встречался с которыми что-то не так...
Я не знаю от чего зависит. Вот на свежую прошил последнюю версию - тоже какие-то отвалы...
Раньше работало.

Разница - в последней версии библа ESPAsyncWenServer была обновлена с версии 1.2.3 на 1.2.4 - потому что 1.2.3 не компилилась под FastLED 3.9.4

Может быть стоит забить на совместимость и вернуться к предыдущей версии... Ткм более 3.9.4 с 8266 все равно не работает...

barkas803, 2024-12-10:
Спасибо за быструю отзывчивость :)

vvip-68, 2024-12-10:
Попробуйте скачать вот эту версию -
https://github.com/vvip-68/LedPanelWiFi/tree/4c2beb03675643aaad91a10c9c51dcaf292a1990

Она до всех изменений адаптации к FastLED 3.9.4 нормально работала
Библиотеки из нее возьмите

barkas803, 2024-12-10:
> Попробуйте скачать вот эту версию - https://github.com/vvip-68/LedPanelWiFi/tree/4c2beb03675643aaad91a10c9c51dcaf292a1990
> 
> Она до всех изменений адаптации к FastLED 3.9.4 нормально работала Библиотеки из нее возьмите

другую версию залили?
Я сегодня часов 5 назад качал, вроде такая же версия :/ могу ошибаться

vvip-68, 2024-12-10:
По ссылке - версия в GIT до изменений под версию FastLED 3.9.4
И библиотеки в ней другие, несовместимые.
Вот поставил ее с FastLED 3.7.0 - никаких падений и дисконнектов....

Но в семой библе FastLED все же исправить время нужно. Иначе будет мерцающий диод...

barkas803, 2024-12-10:
> По ссылке - версия в GIT до изменений под версию FastLED 3.9.4 И библиотеки в ней другие, несовместимые. Вот поставил ее с FastLED 3.7.0 - никаких падений и дисконнектов....
> 
> Но в сaмой библе FastLED все же исправить время нужно. Иначе может быть мерцающий диод...

Залил, всё так же, исправил только в скетче подкл к wifi и матрицу за место 32 поставил 16 =(
Единственное что теперь ещё и с com порта скинул, то есть логи уже не посмотреть в режиме онлайн.

vvip-68, 2024-12-10:
Хм... А я как раз залил эту версию с этими библиотеками - работает без проблем и отвалов....
Ну тогда - не знаю... Попробуйте какую-нибудь другую гирлянду - их много. WLED вон последних версий сильно хвалят...

barkas803, 2024-12-11:
Поставил wled через web сайт, web-морда работает без проблем, наверное там кода меньше, работает контроллер от usb-компа, web-морда не падает и через приложение iOS тоже не падает контроллер. 
Конечно попробую потом с другим контролером например esp32, я за отечественных кодеров :)
Спасибо :)

vvip-68, 2024-12-11:
Ну и отлично. wled, похоже, пишет команда, которая только им и занимается.  Отсюда и быстрое развитие.
Для меня же этот проект - хобби в свободное от работы время, в одного человека, по большей части когда меня что-то в моей гирлянде или настольных часах перестает устраивать и я переписываю / дописываю под себя. Отсюда и все различия...

closed 2024-12-13 by vvip-68