vvip-68/LedPanelWiFi #100: Загрузка строк из внешнего файла настроек (предложение).

Des-75, 2024-12-19:
Здравствуйте.
Хочу предложить сделать загрузку примеров строк для "Бегущей строки" при первом запуске прошивки по аналогии с другими настройками - из внешнего файла.
IMHO было бы удобно.

vvip-68, 2024-12-19:
Хорошо, делайте. Внедрим.

Des-75, 2024-12-19:
Я вас понял :)

closed 2024-12-19 by Des-75

vvip-68, 2024-12-19:
А, вы хотели, чтобы я это делал?
Подумаем... Но не скоро.

Des-75, 2024-12-19:
Я предложил. Потому что сам это вряд ли смогу. Я посмотрел код (сильно звучит, в моём случае), чтобы понять как вы реализовали чтение настроек из внешнего файла. И ничего по коду не понял.
Моя логика такая (по примеру A_DEF_DEVC):
если A_DEF_DEVC определён в "1", то все настройки в самом скетче должны игнориться. А как это происходит, я не понимаю. Ну, т.е. я представляю конструкцию if .. end_if, но не вижу её. И вообще никакой, кроме как подключения файла A_DEF_DEVC.h не вижу. Но, раз всё работает, значит... я ничего не понимаю. 
Вот такого уровня я "погромизд". :)

vvip-68, 2024-12-19:
Никакого там чтения настроек из внешнего файла нет.
Это директивы условной компиляции. Если условие верно - компилятор подгружает указанный файл, который переопределяет настройки, заданные в основном файле a_dev_hard.h. Просто этот файл - с настройками для всех (зачем другим нужны настройки под конкретно мой набор устройств), а в том "внешнем" файле описаны мои устройства.

Тексты точно так же - заданы в файле a_def_lang.h и подцепляются при компиляции из него. Если включен параметр (а по умолчанию он включен) - #define INITIALIZE_TEXTS 1

Des-75, 2024-12-19:
Спасибо, за объяснение. Мне кажется, я понял всю абсурдность своего "предложения": хочешь, бери и правь собственно "внешний" (в моих не верных определениях) файл a_def_lang.h.