vvip-68/LedPanelWiFi #106: Не видит файлы

udiFi, 2024-12-22:
[image]

vvip-68, 2024-12-22:
Загрузите туда файлы. Сейчас эти папки пустые.

closed 2024-12-22 by vvip-68

udiFi, 2024-12-22:
[image]

vvip-68, 2024-12-22:
Вероятно у вас либо новая ArduinoIDE 2.x.x - ее плагины временами глючат и не могут создать пакет для загрузки. Вроде не линукс - на линуксе подобные проблемы были.  Если с 10 попыток загрузить не получится - поставьте Arduino IDE 1.8.19 и ее плагины для загрузки в файловую систему. Они вроде не глючили ни разу.
Ну или используйте отдельные утилиты типа espTools - но я их не пользовал - ничего подсказать не могу.

Вот тут человек как-то решил проблему - свяжитесь с ним.
https://github.com/vvip-68/LedPanelWiFi/issues/102

udiFi, 2024-12-22:
1.8.19, загружал и с платформио и онлайн. Зато на счет красной ошибки узнал https://github.com/me-no-dev/arduino-esp32fs-plugin/issues/3#issuecomment-337079574

vvip-68, 2024-12-22:
Так это понятно, что SPIFFS - ограничение 32
Только там не SPIFFS используется - а LittleFS.
И почему плагин готовит файлы как будто под SPIFFS а не под LittleFS - загадка.
И почему другие плагины (или ээти время от времени) - нормально упаковывают данные файлы...
Вот у меня такая ошибка ни разу не встречалась.
У других - довольно часто - не вы первый...
Они заливали из под 1.8.19 - заливалось. Под 2.3.3 - не всегда.

udiFi, 2024-12-22:
"И почему плагин готовит файлы как будто под SPIFFS а не под LittleFS - загадка." - сам не знаю, может глянуть на скрины что в проекте

https://github.com/vvip-68/LedPanelWiFi/blob/main/setitngs-esp32.png

vvip-68, 2024-12-22:
Эти скрины отражают то, что работает у меня, что проверено.
Можете выбрать - там вариант есть с FAT. Может взлетит. Только все равно файловая система - LittleFS.