vvip-68/LedPanelWiFi #114: Бегущая строка на WIFI Panel

Wod73, 2025-01-07:
Здравствуйте. Собрал панель по вашей схеме на esp32 wroom32. Эффекты работают. Бегущая строка пытается работать, но выводит зеркально. Погода и часы не понятно что выводится. Библиотеки проверил, версии той которые в инструкции указаны. Подскажите в чём может проблема? Если пробовать менять точку подключения (подключено к правому нижнему углу, направление вверх) на верхний правый, текст начинает правильно отображаться.

vvip-68, 2025-01-07:
В скетче нет никаких других регулировок. Все определяет точка подключения и направление из угла. 99.9999% пользователей подключают и настраивают подключение в скетче (или из веб интерфейса) именно так как подключено физически. Ваш случай - первый и единственный. У меня нет ответа на ваш вопрос и я даже представить себе не могу, зная скетч, по какой причине может быть такой эффект как у вас.

Wod73, 2025-01-07:
нет, зигзаг используется. Использую ленту, собирал последовательно по стрелкам на ленте.

vvip-68, 2025-01-07:
И в настройках тоже зигзаг используется? Дайте скриншот страницы Еастройка / Матрица

Wod73, 2025-01-07:
Доэспереминтировался...сейчас в дефолте залью

Wod73, 2025-01-07:
Вот так по дефолту работает.
[image]
[image]
[image]

vvip-68, 2025-01-07:
Что-то внешне это выглядит как будто у вас лента - WS2811? в которой одна "точка" - это три подряд идущих диодов.
ТО есть в ней адресация не отдельными диодами, а тройками - три последовательно идущих диода это одна точка, следующие три - вторая точка и так далее. Такая лента для гирлянды не подойдет.

Проверьте - залейте простой скрипт который в цикле перебирает от 0 до NUM_LEDS-1 диоды, зажигая на 0.5 сек  первый, потом следующий и так жадее. Если будет бежать по одному диоду - ок. Но в вашем случае скорее всего будет бежать группами по три диода. Сначала загорится первые три, потом вторые три и так далее.

vvip-68, 2025-01-07:
Укажите высоту матрицы для эксперимента не 30, а 10.

vvip-68, 2025-01-07:
Укажите высоту 10 - будет у вас матрица 44х10 - нормально работать

Wod73, 2025-01-07:
[image]
[image]

Wod73, 2025-01-07:
Да, так заработало. Наскоком не получилось сделать. Спасибо.

closed 2025-01-08 by vvip-68