vvip-68/LedPanelWiFi #115: Бегущая строка как отдельный эффект

Sh-Renat, 2025-01-07:
Добрый день! У вас самая крутая прошивка, все в ней хорошо, настроек море, веб работает отлично. Очень не хватает бегущей строки как отдельный эффект на своем фоне, а можно и без фона. Это было у Гайвера во вкладке кастом, я после каждого третьего эффекта пускал бегущую строку С новым годом!!!, у вас это только поверх других эффектов. Так как у меня количество светодиодов 12х12 это выглядит кашей в купе с другими эффектами, но как отдельный эффект выглядит нормально и читабельно. Как вариант было бы хорошо пускать отдельный текст после например 5 минут, а можно и после каждого третьего эффекта. Спасибо. Успехов. С новым годом.

vvip-68, 2025-01-07:
Ставите интервал показа бегущей строки 300 секунд - будет вам строка бежать через каждые 5 минут.

Sh-Renat, 2025-01-07:
Я ставил Текст поверх эффектов ВЫКЛ и время через каждые 30 секунд. Нету текста. А когда Текст поверх эффектов ВКЛ то текст бежит поверх эффектов через 30 секунд. Как самостоятельный эффект не работает.

Sh-Renat, 2025-01-07:
Вот палитру фона В я не использовал, если оно будет с ней работать это норм. Проверю.

vvip-68, 2025-01-07:
Ну естественно, если выключаете - то не будет показываться.
Добавьте в текст строки макрос {B#000000} чтобы задать фон текста

Sh-Renat, 2025-01-07:
Только что проверил, работает. Спасибо.

closed 2025-01-08 by vvip-68