vvip-68/LedPanelWiFi #117: Использование ленты с чипом SK9822

Mosrom, 2025-01-27:
Уважаемый разработчик, ещё раз благодарю за такой замечательный проект.

При использовании чипа ленты SK9822 с четырьмя выводами (VCC, GND, DI, CI) какие конкретные изменения нужно произвести в скетче?
Понятно, что во вкладке "a_def_hard.h" в строке "#define LED_CHIP" нужно указать новый тип чипа ленты.
Возможно нужно еще изменить функцию "allocateLeds()" в utility.ino (или utility2.ino), но что конкретно в ней менять просьба подсказать.
Так как появился дополнительный вывод ленты, вероятно нужно добавить строку во вкладку "a_def_hard.h" с указанием значения номера пина на модуле ESP32, куда будет подключен данный вывод ленты.
Если не затруднит, поясните пожалуйста пошагово.

vvip-68, 2025-01-27:
У меня нет опыта с данными лентами.
Но чисто теоретически - вы все правильно сказали.
Изменить тип в LED_CHIP, завести рядом с LED_PIN (data pin) определение для CLOCK_PIN,
и согласно примерам из FastLED для этой ленты в строки инициализации указывается этот CLOCK_PIN

Согласно опять же примерам, для SK9822 формат инициализации следующий:

FastLED.addLeds<SK9822, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);

Это и нужно исправить в allocateLeds()
Можно просто весь этот огромный case выкинуть и заменить конкретно вашим набором, т.к
скетч и веб-интерфейс все равно не рассчитаны на данный тип ленты - там можно из Web-интерфейса менять тольrо назначение DATA_PIN, а для CLOCK_PIN не предусмотрено

Mosrom, 2025-01-27:
Большое спасибо за быстрый ответ, буду пробовать...

Mosrom, 2025-01-30:
Добрый вечер!

Поменял тип ленты и добавил определение для DATA_PIN и CLOCK_PIN. У меня тип модуля ESP32S3 Dev Module.

<pre>
<font color="#5e6d03">#if</font> <font color="#000000">defined</font><font color="#000000">(</font><font color="#000000">ESP32</font><font color="#000000">)</font>

<font color="#5e6d03">#define</font> <font color="#000000">WIDTH</font> <font color="#000000">45</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#95a5a6">&#47;* ширина одного сегмента матрицы для MATRIX_TYPE == 0 или MATRIX_TYPE == 1 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;| для MATRIX_TYPE == 2 ПОЛНАЯ ШИРИНА матрицы &nbsp;&nbsp;*&#47;</font>
<font color="#5e6d03">#define</font> <font color="#000000">HEIGHT</font> <font color="#000000">30</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#95a5a6">&#47;* высота одного сегмента матрицы для MATRIX_TYPE == 0 или MATRIX_TYPE == 1 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;| для MATRIX_TYPE == 2 ПОЛНАЯ ВЫСОТА матрицы &nbsp;&nbsp;*&#47;</font>
<font color="#5e6d03">#define</font> <font color="#000000">MATRIX_TYPE</font> <font color="#000000">0</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#95a5a6">&#47;* тип соединения диодов в сегменте матрицы: 0 - зигзаг, 1 - параллельная, 2 - использовать карту индексов &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;*&#47;</font>
<font color="#5e6d03">#define</font> <font color="#000000">MATRIX_INDEX</font> <font color="#000000">1</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#95a5a6">&#47;* только для MATRIX_TYPE == 2 место расположения массива индексов 0 - в сектче, 1 - в файле &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;| для MATRIX_TYPE == 0 или 1 - поставьте тут 1 *&#47;</font>
<font color="#5e6d03">#define</font> <font color="#000000">CONNECTION_ANGLE</font> <font color="#000000">3</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#95a5a6">&#47;* угол подключения диодов в сегменте: 0 - левый нижний, 1 - левый верхний, 2 - правый верхний, 3 - правый нижний | для MATRIX_TYPE == 2 - не используется &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;*&#47;</font>
<font color="#5e6d03">#define</font> <font color="#000000">STRIP_DIRECTION</font> <font color="#000000">1</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#95a5a6">&#47;* направление ленты из угла сегмента: 0 - вправо, 1 - вверх, 2 - влево, 3 - вниз &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;| для MATRIX_TYPE == 2 - не используется &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;*&#47;</font>

<font color="#5e6d03">#define</font> <font color="#000000">META_MATRIX_WIDTH</font> <font color="#000000">45</font> &nbsp;&nbsp;&nbsp;&nbsp;<font color="#95a5a6">&#47;* количество сегментов в ширину сборной матрицы &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;| для MATRIX_TYPE == 2 - не используется &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;*&#47;</font>
<font color="#5e6d03">#define</font> <font color="#000000">META_MATRIX_HEIGHT</font> <font color="#000000">30</font> &nbsp;&nbsp;&nbsp;<font color="#95a5a6">&#47;* количество сегментов в высоту сборной матрицы &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;| для MATRIX_TYPE == 2 - не используется &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;*&#47;</font>
<font color="#5e6d03">#define</font> <font color="#000000">META_MATRIX_TYPE</font> <font color="#000000">0</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#95a5a6">&#47;* тип сборной матрицы: 0 - зигзаг, 1 - параллельная &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;| для MATRIX_TYPE == 2 [... cut ...] )</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

</pre>

Однако, чего-то не хватает, возникает ошибка, ругается на новые переменные (файл с ошибками во вложении).

Буду очень благодарен, если поможете доработать скетч под данный тип чипа ленты.

И в принципе, в будущем, пожелание - чтобы была возможность выбирать необходимый тип сразу в скетче, включая чипы подобные этому.

[1234.txt](https://github.com/user-attachments/files/18608757/1234.txt)

vvip-68, 2025-01-31:
По доработке в будущем. Это невозможно.
Вот это вот объявление типа ленты - addLeds() с параметрами - в FastLED на языке cи - это шаблон (<TEMPLATE>) который переменные не принимает, только константы - компилятор ругается если попытаться передать сюда переменную. Я не знаю как (и возможно ли) это обойти. Именно поэтому здесь вот этот огромный switch-case который вы удалили, чтобы хоть как-то из приложения настраивать пин подключения ленты и порядок цвета. Если в подобный switch-case оборачивать еще и все возможные комбинации типов лент - скетч будет десятки тысяч строк и скорее всего не влезет в память микроконтроллера. Если вы знаете как в си сделать templates принимающие переменные - welcome - расскажите, исправим.

По ошибкам.
Первое что бросается в глаза - у вас в пути к скетче библиотекам кириллица. 90% что это принесет дополнительные проблемы. Люди уже сталкивались.

В строке 582 | #define DATA_PIN (14U) // пин ленты<br>
Что это за <br> - именно на него ругается компилятор. И далее во всех местах где встречается употребление DATA_PIN соответственно компилятор выдает ошибку. Уберите вообще комментарий в этой строке - // и все что за ним до конца строки. Возможно поможет.

Далее - у вас по всему тексту встречается вот этот вот <br> который ломает весь скетч.
Такое впечатление, что скетч редактировался в каком-то онлайн web-редакторе, который все нормальные переводы строки заменил на HTML-ный  перевод строки<br>. Отсюда и всё сломано. 

В общем - ваш скетч безнадежно испорчен. Скачайте заново архив с проектом и используйте нормальный редактор для внесения изменений.

vvip-68, 2025-01-31:
Еще вариант, что вы мне предоставили лог, снятый с какой-то веб-страницы и на ней вот эти ```<br>``` мешают пониманию того, что происходит - дайте нормальный лог компилятора из IDE. К тому же складывается впечатление, что лог не полный и вначале чего-то не хватает.

Начните с того, что просто уберите кириллицу из пути к файлам - возможно этого будет достаточно.
Или соберите скетч в ArduinoIDE 1.8.19 - она в отличие от 2.3.x к кириллице в путях относится более толерантно...

Mosrom, 2025-01-31:
Скетч был собран в  ArduinoIDE 1.8.19.
Путь теперь без кириллицы, папку переместил.
DATA_PIN проверил, вроде всё корректно, просто до CLOCK_PIN компиляция не дошла, остановилась на DATA_PIN. 
Лог ошибок во вложении.

[image]

[SK9822.txt](https://github.com/user-attachments/files/18613882/SK9822.txt)

Mosrom, 2025-01-31:
> По доработке в будущем. Это невозможно. Вот это вот объявление типа ленты - addLeds() с параметрами - в FastLED на языке cи - это шаблон (`<TEMPLATE>`) который переменные не принимает, только [...]

Не в эту тему, но касательно объема занимаемой памяти скетча в связи с возросшим объемом скетча и оптимизации его работы, может подумать на тем, чтобы исключить из скетча поддержку устаревших плат на базе микроконтроллера ESP8266?

vvip-68, 2025-01-31:
По поводу лога. У меня идей нет на что ругается.
Задал вопрос в форуме - может кто что подскажет.

По поводу исключения ESP8266 - а смысл?
Код генерируемый для esp8266 и для esp32 вообще не пересекается нигде.
В зависимости от того, какой контроллер выбран - компилятор берет либо одни библиотеки / компоненты, либо другие. И вместе они никак и никогда не пересекаются.

vvip-68, 2025-01-31:
Вот умные люди сказали, что вы зря переименовали LED_PIN в DATA_PIN, потому что такое имя уже есть внутри самой библиотеки - возникает конфликт имен. Возможно CLOCK_PIN уже тоже есть определение внутри библиотеки и тоже будет ругаться. Надо назвать по-другому.

[image]

Пусть это будет по-прежнему 
```
#define LED_PIN (14U)
#define CLK_PIN (22U)
```
И тогда в месте на картинке выше - 
```
FastLED.addLeds<SK9822, LED_PIN, CLK_PIN, RGB>(leds, NUM_LEDS);
```

Mosrom, 2025-01-31:
Спасибо за подсказку, наименования пинов везде поменял.
Этот шаг пройден.

Далее ругается на пины подключения ленты:

[image]

[SK9822_1.txt](https://github.com/user-attachments/files/18616080/SK9822_1.txt)

vvip-68, 2025-01-31:
Могу только сказать, что не находит части функций.
Куда они делись? Не знаю.
Либо (случайно) удалили, либо структура скетча нарушена так, что оно не находит нужного.
Например лишняя скобка или отсутствие таковой когда ожидается. 
Нет точки с запятой, дам где ее хочет компилятор
Несоответствие открывающих директив типа #if и нет конечной #endif
Конкретного ничего сказать не могу.

Вероятно вы удалили какую-то лишнюю часть при удалении того большого switch-case

Mosrom, 2025-01-31:
Я поэтому Вас и прошу посмотреть изменения во вкладке utility2.ino и скетч в целом на предмет ошибок, т.к. вы обладаете более глубокими познаниями в языке программирования.

Я испробовал разные варианты, похоже ошибка не в скобках, вылезает ошибка «Invalid pin specified» - т.е. почему-то не найдены определения пинов. 

Посмотрите пожалуйста как будет время. Я думаю для Вас изменить в двух вкладках настройки не составит большого труда.
Очень обидно, матрица на данной ленте собрана, а запустить её не получается, я думал, что будет гораздо проще это сделать.
Готов отправить рабочую версию последней редакции Вашего скетча с библиотеками, настроенную по мой модуль ESP (без последних правок). Либо просто подскажите какие позиции во вкладке utility2.ino нужно убрать, а какие оставить под тип ленты SK9822.

Разумеется донат с меня.

vvip-68, 2025-02-01:
Как я посмотрю ваш скетч - у меня же нет его.
Ошибки в компиляторе - в первую очередь не находит функций. 
Значит где-то потеряна закрывающая скобка - нарушена целостность скетча - он его даже скомпилировать не может - не находит обозначенных функций.

Тут можно запомнить строку с инициализацией addLED которая, восстановить этот исправленный файл, содержащий функцию allocateLEDS из проекта и заново править его.

Можно вернуться к ArduiunoIDE 2.3.4 - там есть функционал сворачивания блоков кода. Свернуть ненужный кусок кода - он соблюдает при сворачивании парность открывающих и закрывающих скобок и удалить целиком. Вернуть на место этот запомненный addLeds().

Про Invalid pin - ну значит указанный вами микроконтроллер не допускает использование пина 22 в качестве CLOCK_PIN - выберите другой.

closed 2025-02-11 by vvip-68

Mosrom, 2025-02-12:
Спасибо за наводку по версии ArduiunoIDE 2.3.4.
Да, была ошибка в части удаленного блока, как Вы и предположили.
В IDE действительно очень наглядно видна связь структуры блоков кода. 
Всё же пошел по пути удаления огромного case, скетч скомпилировался корректно!
Осталось доделать блок управления и испытать матрицу в деле.
По результату отпишусь.

Mosrom, 2025-02-20:
> Спасибо за наводку по версии ArduiunoIDE 2.3.4. Да, была ошибка в части удаленного блока, как Вы и предположили. В IDE 2.3.x действительно очень наглядно видна связь структуры блоков кода. Всё же пошел по пути удаления огромного case, скетч скомпилировался корректно! Осталось доделать блок управления и испытать матрицу в деле. По результату отпишусь.

Супер, матрица работает. Автору ещё раз огромный респект.
Одно пожелание, если добавить ещё один размер шрифта часов
для больших матриц, было бы очень здорово, например 10x14.
У меня матрица на 2640 диодов, конечно на ней часы смотрятся очень куцыми.

[image]