vvip-68/LedPanelWiFi #121: Скетч слишком большой

dmsergeevich63, 2025-02-05:
При компиляции скетча для esp32 не хватает памяти для прошивки, дополнительные модули отключены
`Скетч слишком большой; прочитайте https://support.arduino.cc/hc/en-us/articles/360013825179
Скетч использует 1451817 байт (110%) памяти устройства. Всего доступно 1310720 байт.
Глобальные переменные используют 68708 байт (20%) динамической памяти, оставляя 258972 байт для локальных переменных. Максимум: 327680 байт.`
Используемые настройки платы:
[image]

При указании другого размера памяти устройство успешно прошивается, но нет ота, веб-интерфес постоянно теряет связь, а матрица краснеет, будто с красным фильтром, становится тусклой после смены эффекта. Последнее возможно из-за питания([решение описано, пока не применялось, на 1.13 такое было при высокой яркости](https://github.com/vvip-68/LedPanelWiFi/wiki/%D0%A2%D1%80%D0%B8-%D1%88%D0%B0%D0%B3%D0%B0-%D0%BA-%D1%83%D1%81%D0%BF%D0%B5%D1%85%D1%83.-%D0%9D%D0%B5%D0%BA%D0%BE%D1%82%D0%BE%D1%80%D1%8B%D0%B5-%D0%BE%D1%81%D0%BE%D0%B1%D0%B5%D0%BD%D0%BD%D0%BE%D1%81%D1%82%D0%B8-%D0%BD%D0%B0%D1%81%D1%82%D1%80%D0%BE%D0%B5%D0%BA...)) Пробовал прошивки v.1.14.2025.0107 и main(0160718) от 05.02.2025.
[image]

dmsergeevich63, 2025-02-05:
Содержимое a_def_hard.h: 
`// ****************** ПРОФИЛИ УСТРОЙСТВ *******************
//
// ВНИМАНИЕ!!! 
// Рекомендации по выбору ядра для ESP32 и ESP8266 и платы для сборки в настройках Arduino IDE смотри в комментариях в начале основного файла скетча.
//
// --------------------------------------------------------

#define HOST_NAME  F("WiFiPanel")

#define LANG  'RUS'           // Определение языка параметров, передаваемых в приложение управления или Web-интерфейс 'RUS','ENG' или другой, определенный в файле a_def_lang.h
                              // Доступные языки смотри в файле a_def_lang.h; Можете при необходимости там же добавить свой язык

#define A_DEF_DEVC 0          /* 1 - Настройки профилей ваших устройств вынесены в отдельный файл "a_def_devc.h" */
                              // 0 - Настройки профиля вашего устройства размещены ниже в этом файле
                              // Файл "a_def_devc.h" в комплект не входит, нужно создать, скопировать туда настройки профилей ваших устройств
                              // Пример содержимого файла можно взять из архива public.zip и использовать его в качестве шаблона, изменив настройки в нем
                              
#define A_DEF_PASS 0          /* 1 - Настройки API KEY OpenWeatherMap в отдельном файле a_def_pass.h */
                              // 0 - Настройки API KEY OpenWeatherMap определен ниже
                              // Файл a_def_pass.h в комплект не входит, нужно создать, скопировать туда указанные строки
                              // Пример содержимого файла можно взять из архива public.zip и использовать его в качестве шаблона, изменив настройки в нем

#define INITIALIZE_TEXTS 1    /* Заполнить память текстов бегущей строки текстом из примеров при первичной загрузке прошивки или при смене значения EEPROM_OK, когда */
                              // все настройки сбрасываются в значения по-умолчанию
#define TEXT_SHIFT 1          /* На сколько колонок сдвигать бегущую строку за один раз. По умолчанию - на одну колонку */
                              // Скорость сдвига строки зависит от скорости вывода картинки на матьрицу. На больших матрицах требуется больше времени на вывод одного кадра
                              // Если вам кажется что на максимальной скорости строка все равно движется медленно - укажите параметр TEXT_SHIFT равным 2 мли 3. 
                              // Тогда каждый сдвиг текста будет выполняться сразу на 2 или три колонки. Текст будет двигаться быстрее, но визуально менее плавно

#define USE_TEXT_CACHE 1      // Держать тексты бегущей строки в оперативной памяти (по возможности). 
                              // Кэширование строк в памяти позволит свести к нулю время получения очередной строки для отображения
                              // Загрузка строки из файла требует некоторого времени, что вызывает торможение отображение эффекта перед показом бегущей строки
                              // Однако кэш в памяти уменьшает объем свободной оперативной памяти, что на устройствах с ESP8266 может приводить к частым 
                              // перезагрузкам микроконтроллера при нехватки памяти. Тогда установите значение этого параметра в 0 - не использовать
#if defined(ESP8266)
#define TEXT_CACHE_LINE_SIZE 200   // Макс длина текста бегущей строки, которая будет кэшироваться. Более длинные строки будут по-прежнему загружаться перед использованием из файла
#else
#define TEXT_CACHE_LINE_SIZE 512   // Макс длина текста бегущей строки, которая будет кэшироваться. Более длинные строки будут по-прежнему загружаться перед использованием из файла
#endif

                              

// Варианты профилей для устройства, выполненного на микроконтроллере ESP8266 (NodeMCU, Wemos d1 mini, ...)

#if defined(ESP8266)
#define DEVICE_VARIANT 0      /* Поставьте 0 - если ваше устройство использует только часть из доступных компонентов в схемах с ESP8266 - не полная схема */
                              // Поставьте 1 - если ваше устройство ИСПОЛЬЗУЕТ ВСЕ ВОЗМОЖНЫЕ К ПОДКЛЮЧЕНИЮ компоненты (кнопка, DFPlayer, SD-card, TM1637, управление питанием) - полная схема
                              // Для варианта DEVICE_VARIANT == 1 обязательно должен быть отключен отладочный вывод в Serial, поскольку пины контроллера, связанные с
                              // отладочным выводом (D9 и D10 / RX и TX) заняты под подключение компонент кнопки и индикатора TM1637
                              
                              // Обратите внимание - пины подключения для DEVICE_VARIANT == 0 и DEVICE_VARIANT == 1 - отличаются
                              // Выполняйте сборку устройства в соответственно с выбранной схемой
                              //
                              // При необходимости измените настройки подключении в соответсвии с вариантом вашего устройства:
                              //   для DEVICE_VARIANT == 0 --> строки 135-284
                              //   для DEVICE_VARIANT == 1 --> строки 286-414
#endif

// ===================================================================
// Если вы собираете матрицу из ленты светодиодов или ОДНОЙ готовой заводской матрицы 8x8, 16x16 или 8x32 -,
// укажите ниже в параметрах WIDTH, HEIGHT, MATRIX_TYPE, CONNECTION_ANGLE и STRIP_DIRECTION значения,
// описывающие размеры вашей матрицы, её тип (змейка / параллельная) и угол подключения.
//
[... cut ...]
#if defined(ESP8266)
  #define IN_CMD_SIZE 25
  #define OUT_CMD_SIZE 80
  #define OUT_TPC_SIZE 5
#else
  #define IN_CMD_SIZE 25
  #define OUT_CMD_SIZE 120
  #define OUT_TPC_SIZE 5
#endif

#if defined(ESP8266)
  // В 'NodeMCU' определены D9, D10 и не определены RX, TX
  #ifndef RX
    #define RX  3
    #define TX  1
    #define ESP8266_NODEMCU
  #endif
  // В 'Wemos d1 mini' определены RX, TX и не определены D9, D10
  #ifndef D9
    #define D9  3
    #define D10 1
    #define ESP8266_WEMOS
  #endif
#endif
`

vvip-68, 2025-02-05:
Так а вопрос то в чем? Да, у вас не получится впихнуть невпихуемое.
Либо отказаться от OTA (тогда больше памяти остается под скетч и всё влазит), либо купить микроконтроллер с бОльшим объемом флэш-памяти.

Красная матрица, как вы правильно заметили - недостаточное питание диодов. Разбирайтесь с этим.
Постоянная потеря связи с Web-интерфейсом как правило случается, когда берут первые попавшиеся библиотеки ESPAsyncTCP и ESPAcyncWebServer из интернета, а не из комплекта проекта. Другие причины разрыва связи мне не встречались...