vvip-68/LedPanelWiFi #123: Изменилась анимация сердца

dmsergeevich63, 2025-02-18:
Привет!
Заметил, что анимация сердца изменилась по сравнению с прошивкой GyverPanelWiFi. Она стало реже менять своё положение и двигаться с большим шагом. Странно, но файл Heart.h у прошивок одинаковый. В чем может быть причина? Как вернуть прошлую версию анимации?
Как было раньше: https://disk.yandex.ru/i/GBTNFwLx8Oy0Lw

dmsergeevich63, 2025-02-20:
Да, файл heart.h не менялся, а вот в animation.ino были изменения в 469 строке. Строка относится именно к формированию анимации, изменил её на версию GyverPanelWiFi_v1.13 и сердце стало двигаться как на видео выше, на одну ячейку.
Текущая версия:
https://github.com/vvip-68/LedPanelWiFi/blob/07bac3769df16546af03240b6f6a4dbae53471a0/firmware/LedPanelWiFi_v1.14/animation.ino#L468

Версия GyverPanelWiFi_v1.13:
https://github.com/vvip-68/GyverPanelWiFi/blob/705a81ce8afd3878088429ce6622b317514f6c68/firmware/GyverPanelWiFi_v1.13/animation.ino#L448

Сравнение файлов из версий LedPanelWiFi_v1.14 и GyverPanelWiFi_v1.13:
[image]

vvip-68, 2025-02-20:
Да, похоже в версии 1.13 была ошибка, которую и исправили в версии 1.14.
Ошибка - несоблюдение заданных в описателе анимации временнЫх параметров.

closed 2025-02-27 by vvip-68