vvip-68/LedPanelWiFi #125: Невозможно открыть веб-страницу 192.168.0.100

qulet, 2025-03-07:
информация из последовательного монитора - файлы из каталога данных были загружены......

⸮⸮⸮⸮{R
⸮;>oD#⸮@	⸮⸮⸮DFPlayerError: Таймаут!
MP3 плеер недоступен.

Загрузка текстов бегущей строки:
Нет текстов бегущей строки

Папка '/web' обнаружена.
Cписок файлов web-интерфейса:
   [web]
      [a]
         [i]
            32.svg.gz                       9841
            32c3.svg.gz                     6154
            32c3m.png                      12152
            32s2.svg.gz                     9060
            32s2m.png                       5504
            32s3.svg.gz                     6411
            32s3m.png                       8480
            node.svg.gz                     6325
            w.svg.gz                         979
            wemos.svg.gz                    5018
         [l]
            eng.json.gz                    10852
            esp.json.gz                    11613
            lat.json.gz                    11414
            rus.json.gz                     8881
         tz-eng.json.gz                     5588
         tz-esp.json.gz                     5588
         tz-lat.json.gz                     5588
         tz-rus.json.gz                     6327
      chunk-3IM54G3T.js.gz                  5328
      chunk-3MQREIDX.js.gz                 10070
      chunk-46AJKIN2.js.gz                  4274
      chunk-4HEXTZCI.js.gz                  8514
      chunk-4J3ESD4K.js.gz                  8266
      chunk-5EU4XMF4.js.gz                 13644
      chunk-5QI2BH75.js.gz                  6317
      chunk-5SSDKWJM.js.gz                  9480
      chunk-64ZNTZPC.js.gz                  5579
      chunk-7KR3THG5.js.gz                   323
      chunk-7XSEAOSB.js.gz                  7099
      chunk-AYYGXFO5.js.gz                 16161
      chunk-BVJUBOFL.js.gz                   918
      chunk-CJQS6RUM.js.gz                  3835
      chunk-FJL3U3PQ.js.gz                  5558
      chunk-GTMMLIZQ.js.gz                  7459
      chunk-H2GZLP63.js.gz                  3851
      chunk-HNDY2MV5.js.gz                  2902
      chunk-IZGRH33U.js.gz                  7518
      chunk-JI7WS5WA.js.gz                  5224
      chunk-JOI7FNNI.js.gz                  1769
      chunk-JUPGJJQD.js.gz                 18603
      chunk-JVDRGJXS.js.gz                 31182
      chunk-M6HBMXOX.js.gz                  4320
      chunk-MDLLLR26.js.gz                  8613
      chunk-OMYIPD4K.js.gz                 13825
      chunk-PWLZXKIN.js.gz                  4624
      chunk-PWNYKSMO.js.gz                101185
      chunk-QAFUC2OW.js.gz                  6942
      chunk-QLME2ZFM.js.gz                 21217
      chunk-RFY37QB5.js.gz                 14600
      chunk-SQHOVU6R.js.gz                  4992
      chunk-VAHO6XIR.js.gz                  9884
      chunk-VLFI67PD.js.gz                 50715
      chunk-WGX3LEEA.js.gz                  4641
      chunk-ZFTF2I7M.js.gz                  4148
      favicon.ico.gz                         386
      index.html.gz                         3658
      main-BAOEYW26.js.gz                  22892
      [media]
         m-CIXJEH54.woff2                  59000
         w-OL2QU3PE.svg.gz                   979
      polyfills-EONH2QZO.js.gz             12434
      styles-4OBCSMOR.css.gz               10635

-------------------------------------------
Файловая система: 
Использовано 1056768 из 2072576 байт
-------------------------------------------
Свободно памяти: FM: 30160  Max: 27664  Frag: 8
-------------------------------------------

Подключение к Orange_Swiatlowod_D290 -> 192.168.0.100
Нажмите на кнопку для прерывания подключения
и переход в режим точки доступа
......
WiFi подключен. IP адрес: 192.168.0.100, MAC: C4:D8:D5:2B:9F:35, hostname: WiFiPanel
MDNS - запущен

HTTP server started
Свободно памяти: FM: 28144  Max: 27664  Frag: 2
Переход в авторежим отключен
Режим: Облака
Режим: Зебра
Режим: Вышиванка
Режим: Анимация
Режим: Полосы
Режим: Слайды
Режим: Светлячки
Режим: Камин
Режим: Трафик
Режим: Конфетти

qulet, 2025-03-07:
Мой роутер видит:
IP-адрес 192.168.0.100
МАК: C4:D8:D5:2B:9F:35

vvip-68, 2025-03-08:
1.Если в сети уже есть устройство стем же IP-адресом -да,может не открываться из за конфликта адресов.
2. А роутер тоже находится в сети 192.168.0.xxx?
3. Web-сервер на микроконтроллере довольно слабенький, файлы отдает медленно, иногда загрузка прерывается.
Браузер не откроет страницу, пока не загрузит (и не сохранит в кэш) все файлы. Пробуйте несколько раз,может поможет...

Что в самом браузере в консоли разработчика по F12 на вкладке "Сеть",когда загружаются файлы страницы? Есть ли там ошибки типа "загрузка файла прервана"?

vvip-68, 2025-03-08:
Про браузер выше- вот тут есть ли незагруженные  файлы (с ошибкой загрузки)

[image]

qulet, 2025-03-08:
Спасибо за ваш ответ.
Я снова загрузил программу и данные, не вводя имя сети и пароль. Страница в режиме «точки доступа» работает нормально

vvip-68, 2025-03-09:
Потеряете синхронизацию времени с интернета (и время будет только если его с телефона установить как текущее) и получение текущей температуры, если она вам нужна.

А что, если из режима точки доступа в Web-приложении на вкладке Настройки / Сеть / установить имя сети и пароль подключения к ней?

vvip-68, 2025-03-09:
Вы же с Польши? Вам польский язык в интерфейсе не нужен?

vvip-68, 2025-03-09:
А,вот еще вспомнил предположение - проверьте в настройках своего роутера не стоит ли галочка "Изоляция клиентов" (Client isolation) - если она включена- каждое устройство в сети пожет отсылать данные в /из интернета, но между собой общаться не могут - изолированы. Снимите эту галочку.

qulet, 2025-03-09:
Да, я из Польши, хотелось бы, чтобы в интерфейсе и при отправке текста были польские буквы. У меня Funbox Orange 6 и я не вижу такой функции, как изоляция клиента. У меня FluidNC загружен на ESP32. Там подключение сайта работает без проблем.
У меня есть еще один вопрос. У LedPanelWiFi-main есть WEB-каталог, для чего он нужен?

qulet, 2025-03-09:
Я ввел имя сети и пароль из веб-интерфейса точки доступа, выбрал там DHCP и сохранил. соединение с ПД было потеряно. и я не могу подключиться к сайту 192.168.0.100.
PRTSCR F12 /сеть

[image]

[image]

qulet, 2025-03-09:
Чтобы подключиться к ESP, мне придется заново загрузить программу, удалив все данные.

qulet, 2025-03-09:
[image]

vvip-68, 2025-03-09:
> Да, я из Польши, хотелось бы, чтобы в интерфейсе и при отправке текста были польские буквы. У меня Funbox Orange 6 и я не вижу такой функции, как изоляция клиента. У меня FluidNC загружен на ESP32. [...]

Ну это собственно и есть папка с файлами Web-интерфейса. Ее нужно загрузить в микроконтроллер. Но если вы подключаетесь к веб-интерфейсу через точку доступа - значит файлы вы залили на микроконтроллер.

vvip-68, 2025-03-09:
> Я ввел имя сети и пароль из веб-интерфейса точки доступа, выбрал там DHCP и сохранил. соединение с ПД было потеряно. и я не могу подключиться к сайту 192.168.0.100. PRTSCR F12 /сеть

Ну если вы включили DCHP- совсем не факт, что у устройства останется адрес 192.168.0.0 - скорее всего роутер выдал ему новый адрес. Какой - можно посмотреть в мониторе порта при старте микроконтроллера. Еще один способ- при работающем устройстве быстро 4 раза нажать на кнопку- тогда контроллер выведет текущий IP полученный от роутера на экран.

vvip-68, 2025-03-09:
> Чтобы подключиться к ESP, мне придется заново загрузить программу, удалив все данные. Serial Monitor: Подключение к Orange_Swiatlowod_D290 -> 192.168.0.100 Нажмите на кнопку для прерывания подключения и переход в режим точки доступа ...... WiFi подключен. IP адрес: 192.168.0.100, MAC: C4:D8:D5:2B:9F:35, hostname: WiFiPanel MDNS - запущен
> 
> HTTP server started Свободно памяти: FM: 28144 Max: 27664 Frag: 2

Ну вот - пишет, что адрес 192.168.0.100- если файлы веб-интерфейса загружены в микроконтроллер - к нему должно быть возможным подключиться через браузер по этому адресу.

А что - с компьютера ping на этот адрес проходит - есть ответ от контроллера?

vvip-68, 2025-03-09:
> Да, я из Польши, хотелось бы, чтобы в интерфейсе и при отправке текста были польские буквы.

У меня есть квам коммерческое предложение.
Я добавляю в прошивку поддержку польских букв и для веб-интерфейса даю инструкции как перевести веб-интерфейс на польский. Для этого вам, как носителю языка придется перевести строки в указанных файлах с русского или английского на польский и файлы отправить мне - я включу их в следующий релиз. Выпотом обновите прошивку и зальете новую файловую систему и будет интерфейс на польском. Думаю, что если я буду переводить гугл-переводчиком это скорее всего будет не совсем корректно. Или могу перевести так,а вы потом исправите...

Что я прошу взамен. В связи с известными событиями платежи из России в интернет магазины (речь идет об Esty) заблокированы. Я не могу купить там нужное мне. Мне бы хотелось, чтобы вы купили это по возможности и переслали файлы мне. Речь идет о 5-10 долларов США. По тем же причинам - возместить данную сумму вам я не смогу,если только у вас нет карточки на которую принимаются денежные переводы из России :)

Речь идет вот об этой позиции - https://www.etsy.com/listing/1729501592/podaronaa-korobka-s-lazernoj-rezkoj-i
Текущая отображаемая цена - $3.94

[image]

Есть еще одна позиция, но наверное с моей стороны это будет запредельной наглостью - поэтому можно и нет...
https://www.etsy.com/listing/1509885968/dekorativnaa-derevannaa-podaronaa - эта сейчас $5.04

[image]

Был бы премного благодарен...

vvip-68, 2025-03-09:
В интернете очень мало информации по "Funbox Orange 6"
А он точно находится в сети 192.168.0.xxx? Какой у него самого адрес- как вы из браузера к нему подключаетесь?
Он точно поддерживает сеть WiFi 2.4ГГц в которой работает контроллер.Если только 5ГГц- контроллер не сможет подключиться.

Судя по скриншоту F12 - браузер не может ничего вообще получить с этого адреса. Ping с компа на микроконтроллер проходит? Ответ есть?

vvip-68, 2025-03-09:
[image]

Вот на этом фото роутера написано - что его сеть по умолчанию - 192.168.**1**.xxx
Чтобы микроконтроллер мог быть доступен в сети через роутер - он тоже должен находиться в сети 192.168.**1**.xxx - а не 192.168.**0**.xxx как задано по умолчанию.  Нужно или в настройках скетча изменить адрес микроконтроллера по умолчанию или все же использовать DHCP

qulet, 2025-03-09:
Я готов купить вышеуказанный товар и отправить его по указанному адресу.
Фанбокс:
IPv4-адрес 83.11.135.173
Тип доступа FTTH
 Мой FluidNC ESP32 работает по адресу 192.168.1.22
С DHCP адрес просто 0.0.0.0?

vvip-68, 2025-03-09:
Если ставите галочку DHCP - оставьте поле IP адреса вообще пустым.
Какой адрес будет получен - видно в мониторе порта при старте скетча или по 4-хкратному нажатию кнопки, подключенной к контроллеру.

За покупку - огромное спасибо.
Поддержку польского - добавлю на неделе.
Файлы которые нужно перевести - напишу где что и как.

qulet, 2025-03-09:
При покупке я могу предоставить вам адрес, по которому вы сможете забрать товар.
Пожалуйста, отправьте его мне, и я оформлю покупку сегодня за 5 долларов.
Я отправил электронное письмо с адреса:
quletqwert@gmail.com

closed 2025-03-11 by vvip-68