vvip-68/LedPanelWiFi #126: Бегущая строка не двух разных лентах

SSergeyS-34, 2025-03-09:
При изготовлении гирлянды 18х12 оказалось, что два куска не GRB а RGB. Подключил их в конце, посадил на пин2 и исправил это в WEB интерфейсе, всё работает правильно. Но когда идёт бегущая строка то в этом месте где две линии RGB "месиво" световое и когда строка уже пости прошла там всё равно хаотично светятся диоды. Такое впечатление, что бегущая строка идет отдельно и по пину D1 и по пину D2. Настройки матрицы 18х12, начало внизу с лева и вверх. Настройка "оборудования" GRB D1- 0-192 и RGB D2 - 0-24. Как можно это исправить?

vvip-68, 2025-03-09:
Так у вас в настройках диапазоны перекрываются. Нужно на D1 указать номера первого отрезка, на D2 - второго.
Типа для D1 - с 0 192 штуки, для D2 - с 193 и 24 штуки

SSergeyS-34, 2025-03-09:
Огромное спасибо! Всё получилось.

closed 2025-03-09 by vvip-68