vvip-68/LedPanelWiFi #127: Не создается точка доступа PanelAp.

Immanior, 2025-03-11:
Доброго времени суток.  Пишу от безысходности уже.
Решил увеличить размер матрицы с 12ч10 на 16x13 и заодно сделать прозрачные соединительные провода. Гирлянду распаял заново. Но столкнулся с большой (для меня проблемой).
При подключении платы к гирлянде моргает из 208 светодиодов только 120. Причем именно с 1 по 120. Если взять с 121 по 208 и поставить эти столбики в начало гирлянды – они там работают.
При попытке перепрошить микроконтроллер на обновленный размер – данные не сохраняются в него.
После перезагрузки сеть PanelAP почему-то не создается. Создаётся сеть Гусь. Почему именно она – я не знаю. Оставлял управление ребенку на новый год. После нового года забрал плату в командировку и готовлюсь к новому году заранее.
 Помогите! Что делаю не так?
Что значит "ПЕРЕД ЗАГРУЗКОЙ ПРОШИВКИ В МИКРОКОНТРОЛЛЕР ИЗМЕНИТЕ ЗНАЧЕНИЕ КОНСТАНТЫ EEPROM_OK В ФАЙЛЕ a_def_soft.h НА ДРУГОЕ" Какое другое?

### WiFiPanel v.1.14.2023.0602
Контроллер: 'NodeMCU'
Версия ядра: ESP8266 v3.1.2
Версия EEPROM: 0xA5
Host: 'WiFiPanel'

Инициализация файловой системы... 
Использовано 1671168 из 2072576 байт

Нет конфигурации сети: SSID не задан
Нет конфигурации сети: пароль не задан

Матрица: плоская 12x10
Адресация: по подключению
  Угол:   Направление:   Тип: зигзаг
Вывод на ленту:
  Линия 1 PIN=D2, START=0, COUNT=120

Доступные возможности:
+ Язык интерфейса: RUS
+ Бегущая строка: шрифт 5x8
+ Кнопка управления: тактовая PIN=D4
+ Синхронизация времени с сервером NTP
- Управление питанием
- Получение информации о погоде
+ Управление через Web-канал
- Групповая синхронизация по протоколу E1.31
- Дополнительный индикатор TM1637
+ Эффекты Jinx! с SD-карты CS=D8, CLK=D5, MISO=D6, MOSI=D7
- Поддержка MP3 Player

Выделение памяти для LEDS   : 29504 -> 29136 -- 368
Выделение памяти для OVERLAY: 29136 -> 28768 -- 368

Выбранные эффекты и их порядок: 
   01. Радуга
   02. Спектрум
   03. Звёзды
   04. Трафик
   05. Дождь
   06. Снегопад
   07. Синусы
   08. The Matrix
   09. Плазма
   10. Цветной шум
   11. Водоворот
   12. Огонь
   13. Шарики
   14. Звездопад
   15. Мерцание
   16. Светлячки
   17. Змейка
   18. Лабиринт

Отключенные эффекты: 
   Часы
   Лампа
   Кубик
   Пейнтбол
   Конфетти
   Облака
   Лава
   Радужные переливы
   Павлин
   Зебра
   Шумящий лес
   Морской прибой
   Смена цвета
   Циклон
   Северное сияние
   Тени
   Тетрис
   Арканоид
   Палитра
   Вышиванка
   Камин
   Стрелки
   Анимация
   Погода
   Узоры
   Рубик
   Штора
   Слайды
   Рассвет
   SD-Карта

Поиск файлов карт индексов...
Нет файлов карт индексов

Инициализация SD-карты...
SD-карта подключена.
Поиск файлов эффектов Jinx!..
Папка с эффектами /12x10 не обнаружена.
Эффекты Jinx! на SD-карте не обнаружены

Найденные анимации:
   Сердце	8x8

Поиск слайдов на FS в папке '12p10'...
Поиск слайдов на FS в папке '16p16'...
Слайды не найдены. Будут использованы картинки эффекта 'Погода'

Папка '/web' обнаружена.
Cпискок файлов web-интерфейса:
   [web]
      MI.122e921fbc0a837a.woff2            59000
      [assets]
         [images]
            esp32.png                       7978
            nodemcu.png                    10874
            wemos.png                      12207
         [langs]
            eng.json.gz                     8596
            esp.json.gz                     8844
            rus.json.gz                     6712
      favicon.ico.gz                         386
      index.html.gz                         1413
      main.e44c7c6a45b2d807.js.gz         337831
      pf.dcddd532a07c6b2d.js.gz            11979
      runtime.7882e08c7503fb34.js.gz         638
      styles.7e4222ebe23cabfb.css.gz       13020
      wallet.f3b77cb2a7aa9496.png.gz        1169
Свободно памяти: 28616
Создание точки доступа Гусь
..........
Не удалось создать WiFi точку доступа.
Создание точки доступа Гусь
..........
Не удалось создать WiFi точку доступа.
HTTP server started
Свободно памяти: 25568
Переход в авторежим отключен
Режим: Выключено

vvip-68, 2025-03-11:
Ну так то размер матрицы настраивается в Web-интерфейсе.
Если хотите именно в скетче- нужно заливать прошивку с полнойочисткой памяти.

Как по мне -так в Web указать размер проще

[image]

[image]

Immanior, 2025-03-11:
Огромное вам спасибо за ответ!
_Ну или в настройках компиляции в меню "Инструменты" выбрать опцию "Erase all content"для стирания всей памяти перед очередной загрузкой скетча_
Это у меня выставлено значение. Несколько раз проверил - не стирает почему-то. 

_Ну тут два варианта - или ваш ребенок переименовал имя точки доступа с PanelAP на этого самого Гуся,
либо микроконтроллер подключается к локальной сети и в случае успешного подключения точку доступа он не создает (если не указано принудительное создание даже в случае успешного подключения к сети), а Гусь - это вообще что-то постороннее_
Ребенок действительно переименовал сеть в веб интерфейсе. Сознался. Зачем? Руки чесались у программиста. Локальной сети с таким именем у меня не было.
 Вот что пишет в нижнем окне при перепрошивке МК.
```
[LittleFS] data    : E:\LedPanelWiFi-main\LedPanelWiFi-main\firmware\LedPanelWiFi_v1.14\data
[LittleFS] size    : 2024
[LittleFS] page    : 256
[LittleFS] block   : 8192
/16p13/airplan.p
/16p13/arrow-dn.p
/16p13/arrow-lt.p
/16p13/arrow-rt.p
/16p13/arrow-up.p
/16p13/ball.p
/16p13/bird.p
/16p13/butterfly.p
/16p13/cat.p
/16p13/clock.p
/16p13/comet.p
/16p13/cookman1.p
/16p13/elka.p
/16p13/fish.p
/16p13/fox.p
/16p13/fox1.p
/16p13/frog.p
/16p13/frog2.p
/16p13/frog3.p
/16p13/gift.p
/16p13/hand.p
/16p13/heart.p
/16p13/jinn.p
/16p13/kaktus.p
/16p13/karate.p
/16p13/lebedev.p
/16p13/man2.p
/16p13/mario.p
/16p13/minion.p
/16p13/moon.p
/16p13/moroz1.p
/16p13/mouse.p
/16p13/night_town.p
/16p13/parrot.p
/16p13/pigeon.p
/16p13/rocket.p
/16p13/smile1.p
/16p13/smile2.p
/16p13/smile3.p
/16p13/smile4.p
/16p13/smile5.p
/16p13/snowman.p
/16p13/spiderman.p
/16p13/squares.p
/16p13/star1.p
/16p13/termo.p
/16p13/test.p
/16p13/weather.p
/16p13/winter.p
/16p13/zayka.p
/16x13/Fireworks.out
/16x13/Mill.out
/16x13/Psycho.out
/16x13/Stars.out
/web/assets/images/esp32.png
/web/assets/images/nodemcu.png
/web/assets/images/wemos.png
/web/assets/langs/eng.json.gz
/web/assets/langs/esp.json.gz
/web/assets/langs/rus.json.gz
/web/favicon.ico.gz
/web/index.html.gz
/web/main.e44c7c6a45b2d807.js.gz
/web/MI.122e921fbc0a837a.woff2
/web/pf.dcddd532a07c6b2d.js.gz
/web/runtime.7882e08c7503fb34.js.gz
/web/styles.7e4222ebe23cabfb.css.gz
/web/wallet.f3b77cb2a7aa9496.png.gz
[LittleFS] upload  : C:\Temp\arduino_build_549059/LedPanelWiFi_v1.14.mklittlefs.bin
[LittleFS] address : 0x200000
[LittleFS] reset   : --before default_reset --after hard_reset
[LittleFS] port    : COM11
[LittleFS] speed   : 921600
[LittleFS] python   : C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe
[LittleFS] uploader : C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\tools\upload.py

esptool.py v3.0
Serial port COM11
Connecting......
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 8c:aa:b5:0c:d4:bd
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 2072576 bytes to 628829...
Writing at 0x00200000... (2 %)
Writing at 0x00204000... (5 %)
Writing at 0x00208000... (7 %)
Writing at 0x0020c000... (10 %)
Writing at 0x00210000... (12 %)
Writing at 0x00214000... (15 %)
Writing at 0x00218000... (17 %)
Writing at 0x0021c000... (20 %)
Writing at 0x00220000... (23 %)
Writing at 0x00224000... (25 %)
Writing at 0x00228000... (28 %)
Writing at 0x0022c000... (30 %)
Writing at 0x00230000... (33 %)
Writing at 0x00234000... (35 %)
Writing at 0x00238000... (38 %)
Writing at 0x0023c000... (41 %)
Writing at 0x00240000... (43 %)
Writing at 0x00244000... (46 %)
Writing at 0x00248000... (48 %)
Writing at 0x0024c000... (51 %)
Writing at 0x00250000... (53 %)
Writing at 0x00254000... (56 %)
Writing at 0x00258000... (58 %)
Writing at 0x0025c000... (61 %)
Writing at 0x00260000... (64 %)
Writing at 0x00264000... (66 %)
Writing at 0x00268000... (69 %)
Writing at 0x0026c000... (71 %)
Writing at 0x00270000... (74 %)
Writing at 0x00274000... (76 %)
Writing at 0x00278000... (79 %)
Writing at 0x0027c000... (82 %)
Writing at 0x00280000... (84 %)
Writing at 0x00284000... (87 %)
Writing at 0x00288000... (89 %)
Writing at 0x0028c000... (92 %)
Writing at 0x00290000... (94 %)
Writing at 0x00294000... (97 %)
Writing at 0x00298000... (100 %)
Wrote 2072576 bytes (628829 compressed) at 0x00200000 in 17.1 seconds (effective 971.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```
Далее мне нужно нажать клавишу Reset на моей платке D1miniMicro USB?
Увы первый раз зашивал года 3 назад. И не вспомню. Может проще новую плату купить?

vvip-68, 2025-03-12:
1. Никогда не сталкивался с таким,чтобы не стирало. Тут ничего тогда не подскажу.
2. То что выше-это не перепрошивка МК. Это загрузка файловой системы в микроконтроллер. А это совсем другое.
3. Обычно после завершения прошивки или загрузки файловой системы микроконтроллер сам перезапускается.У меня только один ESP32S3 требует нажатия на кнопку Reset/ Впрочем - нажатие на нее после прошивки / загрузки - ничего плохого не сделает.
4. Плату новую стоит купить в плане,чтобы была запасная / подменная. А так и эта вполне себе нормально работает.

Immanior, 2025-03-12:
1. Посыпаю голову пеплом.
2.  Понял.
3. У меня почему-то МК не перезапускается. Раньше точно перезапускался. Может в этом и есть проблема.
4. Размеры матрицы настроить не смог. Так как не могу зайти в веб интерфейс. Создается этот гусь - его нигде не видно.
PanelAP тоже нет.

vvip-68, 2025-03-12:
Тут какое-то противоречие - "создается этот гусь" - и "его нигде не видно".
А как вы про него тогда узнали?
Понятно что если создается гусь - то PanelAP не будет фигурировать. Точнее гусь это и есть PanelAP.
Вам шашечки или ехать? Вы можете подключиться к этому гусю?

Immanior, 2025-03-12:
Да. подключиться к этой непонятной сети тоже не могу. Ее не видят ни мой комп, ни мой телефон

vvip-68, 2025-03-12:
Ну, видимо,имя точки доступа должно быть латиницей, иначе микроконтроллер не может создать ее.
Об чём прямо и пишет.
Тогда единственный способ-перепрошить микроконтроллер с полной очисткой памяти.

Immanior, 2025-03-12:
Чтобы перепрошить с полной очисткой памяти я:
0. Подключаю абсолютно голую плату к компу (отпаяв все кнопки и флэшки)
1. В 7 строчке кода 
#define EEPROM_OK 0x67 // Флаг, показывающий, что EEPROM инициализирована корректными данными
выставил другое число. ( было 0xA5)
2. Вкладка инструменты в строчке Erase flash выставил all flash contents.
3. Нажимаю ESP8266 LittleFS Data Upload.
жду 
4. Запускаю монитор порта и вижу.
Не удалось создать WiFi точку доступа.
HTTP server started
Свободно памяти: 25344
Переход в авторежим отключен
Режим: Выключено

Выкидывать плату?

vvip-68, 2025-03-12:
Я только не понимаю,почему вы вместо того,чтобы скомпилировать и загрузить прошивку все время просто загружаете в нее файловую систему? Так - ничего не поменяется. А микроконтроллер думаю рабочий - не нужно его выкидывать.

vvip-68, 2025-03-12:
https://github.com/vvip-68/LedPanelWiFi/wiki/Пошаговая-инструкция-подготовки-среды-и-сборка-проекта#шаг-9

Immanior, 2025-03-13:
Благодарю. Был без интернета - не мог проверить. Сейчас пройдусь с самого начала по вашему совету.

vvip-68, 2025-03-17:
Если бы вы еще в комментариях в главном файле прочитали какие версии ядра и какие версии библиотек нужны для проекта - вообще б замечательно было.
Спойлер - с первыми попавшимися версиями , найденными где-нибудь в интернете - проект работать не будет. 
Именно это у вас и происходит. Вместо того, чтобы взять ТРЕБУЕМУЮ проектам версию в данном случае FastLED, вы скачали  и установили первую попавшуюся...

Immanior, 2025-03-17:
> A строке поиска наберите "FastLED" Выберите блок "FastLED by Daniel Garcia"
> 
> Выберите версию библиотеки 3.6.0
> Нажмите кнопку "Установка", дождитесь завершения установки.
> Должна появиться надпись, подтверждающая что вы установили правильную (3.6.0!!!) версию библиотеки

Я сделал что-то не так??

closed 2025-03-17 by Immanior