vvip-68/LedPanelWiFi #128: плохо отображается web интерфейс

pashkatara, 2025-03-17:
помогите в какую сторону мне смотреть, а так проект огонь, давно что подобное хотел повторить для дома

[image]

[image]

vvip-68, 2025-03-17:
Ну у вас шрифт не загрузился,который иконки отображает.
Шрифт находится в папке с файлами веб-страницы , файл называется m-CIXJEH54.woff2
Находится в папке sata/web/media
Это странно. Потому что в версиях этого года у этого файла высший приоритет и пока он не загрузится страница вообще не должна бы отображаться.

Проверьте, что этот файл есть в файловой системе в папке web - она распечатывается в мониторе порта при старте микроконтроллера
Попробуйте несколько раз подряд обновить страницу в браузере через Ctrl-F5 и/или Ctrl-Shift-R
Файл должен загрузится в кэш браузера
Откройте панель разработчика в браузере по кнопке F12 и перейдите во вкладку Сеть
Обновите страницу - убедитесь что в консоли браузера нет файлов с ошибками вообще и что файл шрифта тоже загрузился.

[image]

pashkatara, 2025-03-17:
[image]
в папке media есть

pashkatara, 2025-03-17:
[image]

pashkatara, 2025-03-17:
[image]

pashkatara, 2025-03-17:
[image]

pashkatara, 2025-03-17:
[image]
[image]

pashkatara, 2025-03-17:
[image]

pashkatara, 2025-03-17:
вроде загружал по инструкции

pashkatara, 2025-03-17:
[image]

pashkatara, 2025-03-17:
[image]

pashkatara, 2025-03-17:
[image]
[image]
[image]

pashkatara, 2025-03-17:
[image]

vvip-68, 2025-03-17:
Скачайте лучше проект с репозитория.
На представленных скриншотах у вас вообще лишние папки которых быть не должно - папка src например. И еще ряд файлов, которых в папке data быть не должно.  Понятия не имею откуда вы  это взяли - но структура файловой системы (а может и не только ее) - испорчена.
Я не знаю- может быть там не только лишние файлы, но и каких-то не хватает.

pashkatara, 2025-03-17:
ооо вроде появилось

pashkatara, 2025-03-17:
[image]

pashkatara, 2025-03-17:
Спасибо большое, что помогли!!!

closed 2025-03-17 by vvip-68